tinygo-org/tinygo #4519: tinygo flash may not work on Sequoia (macOS 18)

Senoue, 2024-10-12:
Hi. 
I wonder if tinygo flash can't be used on Sequoia (macOS 18).
In my case,

```
failed to flash /var/folders/m9/xxxx/T/xxx/main.uf2: unable to locate any volume: [RPI-RP2]
```

Is everyone okay?

Senoue, 2024-10-15:
sorry. It was macOS 15 instead of 18

sago35, 2024-10-16:
The error is being output from here, and we can predict that the cause lies in `findFATMounts()`. Since I don't have a macOS system, I need someone's help.

https://github.com/tinygo-org/tinygo/blob/v0.33.0/main.go#L994-L1018

aykevl, 2024-10-16:
I have MacOS (though I don't use it much) but am not yet on Sequoia. Kinda hesitant to upgrade, am afraid it will break stuff (last upgrade didn't exactly go very smoothly).

Senoue, 2024-10-17:
Thank you, 
when I checked the mount point on Sequoia, it was mounted as NO NAME in /Volumes, 
so I added NO NAME to targets/rp2040.json

```
"msd-volume-name": ["RPI-RP2", "NO NAME"],
```

I found out that tinygo flash can be done!

Senoue, 2024-12-03:
I had the same problem with two Macs and a friend's environment.

zwindler, 2025-01-31:
I have the issue as well

I installed tinygo as a brew package, and I'm not sure where I can find the "targets/rp2040.json" file with this mean of installation

Can you give a bit more details on how you solved it @Senoue?

I'll try to fix it myself with the information you (kindly) gave in the meantime and report if I find on my own :)

zwindler, 2025-01-31:
Ok, found it. For future readers

When you install with homebrew, the config file is here

```
/opt/homebrew/Cellar/tinygo/0.35.0/targets
```

(Replace 0.35.0 with your version)

And also don't forget to read "Additional Requirements for Microcontrollers" from https://tinygo.org/getting-started/install/macos, or else the rpi2040 will unmount (with an error message)

I still have the issue though, but I'm progressing xD

```
failed to flash /var/folders/dr/vtsgwt8x1ll6bdktx9gkb8p40000gp/T/tinygo89491772/main.uf2: unable to locate any volume: [RPI-RP2,NO NAME]
```

sago35, 2025-02-01:
> /opt/homebrew/Cellar/tinygo/0.35.0/targets

You can find this location by running `tinygo env` or `tinygo env TINYGOROOT`.

> failed to flash /var/folders/dr/xxx/T/tinygo12345/main.uf2: unable to locate any volume: [RPI-RP2,NO NAME]

For the `rp2040` target, the first flash must always be done in **bootloader mode**.  
So, press and hold the **BOOTSEL** button while resetting, then run `tinygo flash`.

mi-bear, 2025-03-23:
I’ve also been struggling with the same issue from time to time.
Right now, I’m looking into possible solutions with help from @sago35.
I might even be able to submit a fix as a pull request…?

closed 2025-08-14 by deadprogram