tinygo-org/tinygo #4526: net/http/httptest not implemented

dkegel-fastly, 2024-10-16:
The very first roadblock in testing libopenapi with tinygo is net/http/httptest not implemented.

```
$ git clone git@github.com:pb33f/libopenapi.git
$ tinygo test ./...
../../go/pkg/mod/github.com/stretchr/testify@v1.9.0/assert/http_assertions.go:6:2: package net/http/httptest is not in std (/Users/dkegel/Library/Caches/tinygo/goroot-9951da0529238c9063c0f528b787a287b032b47730b2310d6d23558a64a022f9/src/net/http/httptest)
```

dgryski, 2024-10-18:
We probably need some hack in https://github.com/tinygo-org/tinygo/blob/release/loader/goroot.go to merge https://github.com/tinygo-org/net and https://github.com/golang/go/tree/master/src/net/http/httptest (and probably figure out what the strategy will be for the rest of the packages under `net`..)

dgryski, 2024-10-31:
Solved by https://github.com/tinygo-org/net/pull/34

deadprogram, 2025-02-17:
Just merged into `dev` and tagged this issue with "next release"

deadprogram, 2025-03-04:
Closed as part of v0.36.0

closed 2025-03-04 by deadprogram