tinygo-org/tinygo #4527: nRF52: a single interrupt activation doesn't work

perttierkkila, 2024-10-17:
TLDR: nRF52 blinky example doesn't work when serial is disabled.

Env
> tinygo version 0.33.0 darwin/arm64 (using go version go1.22.5 and LLVM version 18.1.2)

This is a pretty wild bug which goes down to interrupt enable and this one line https://github.com/tinygo-org/tinygo/blob/release/src/runtime/runtime_nrf.go#L61
> intr.Enable()

When this line is replaced code from Enable() with constant value, IRQ works properly. Only difference is internal irq.num is not used. Enabling any other IRQ doens't do the trick.
> arm.EnableIRQ(uint32(nrf.IRQ_RTC1))

And the weird part, if application enables ANY (for example IRQ num 240 which doesn't exist), IRQ_RTC1 starts working also. Due this, default settings with serial enabled, keeps IRQ_RTC1 working properly and hides a bug.

At least 0.31 worked properly as I have used serial=none setting a long time. Please keep older versions at brew to allow easily allow switching version. Now it is pretty PIA.

perttierkkila, 2024-10-31:
Same problem persists with 0.34.0.
`tinygo version 0.34.0 darwin/arm64 (using go version go1.22.8 and LLVM version 18.1.2)`

Using pca10040-board with blink-code
```
package main

import (
	"machine"
	"time"
)

func main() {
	machine.LED.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		machine.LED.Set(!machine.LED.Get())
		time.Sleep(300 * time.Millisecond)
	}
}
```

When using normal pca10040-target, blinking works. But when modifying pca10040.json and disabling serial, nothing happens anymore. I tracked earlier the reason to IRQ enabling.
```
	"serial": "uart",
```
=>
```
	"serial": "none",
```

perttierkkila, 2024-10-31:
Tested also building images at CI-server using official docker-image. Same problem still exists.
`tinygo version 0.33.0 linux/amd64 (using go version go1.23.0 and LLVM version 18.1.2)`

aykevl, 2024-10-31:
I can reproduce this issue on the pca10040 (after manually setting HasLowFrequencyCrystal to false since the crystal seems to be broken on my board - it shouldn't affect behavior otherwise). I can't seem to reproduce this on a nrf52840. Maybe it's a hardware bug? But it's weird that this only started misbehaving recently.

aykevl, 2024-10-31:
This also happens with `-serial=rtt`, which is useful since it's a way to actually see what's happening:

```
~/src/tinygo/tinygo$ tinygo flash -target=pca10040 -serial=rtt -monitor examples/serial
[...programmer output...]
Connected to . Press Ctrl-C to exit.
hello world!
sleep: 32768
waiting on wfe...
```

aykevl, 2024-10-31:
Found it! See: https://github.com/tinygo-org/tinygo/pull/4572

perttierkkila, 2024-10-31:
Wohoo, excellent! 🎉

closed 2024-11-07 by perttierkkila