tinygo-org/tinygo #4545: Error during compilation

lasarocamargos, 2024-10-25:
Opened an issue against the llvm-project (llvm/llvm-project#113672), but opening it here as well for awareness. Please close if not approapriate

aykevl, 2024-10-25:
I cannot reproduce this issue.

Please provide a full reproducer that shows this problem. Ideally a small but complete code sample (including go.mod and go.sum). Also please share the output of `tinygo version`.

lasarocamargos, 2024-10-25:
The btree.Set example in https://github.com/tidwall/btree?tab=readme-ov-file#example is a repro, that is, this code 

```go
package main

import (
	"fmt"
	"github.com/tidwall/btree"
)

func main() {
	// create a map
	var users btree.Map[string, string]

	// add some users
	users.Set("user:4", "Andrea")
	users.Set("user:6", "Andy")
	users.Set("user:2", "Andy")
	users.Set("user:1", "Jane")
	users.Set("user:5", "Janet")
	users.Set("user:3", "Steve")

	// Iterate over the maps and print each user
	users.Scan(func(key, value string) bool {
		fmt.Printf("%s %s\n", key, value)
		return true
	})
	fmt.Printf("\n")

	// Delete a couple
	users.Delete("user:5")
	users.Delete("user:1")

	// print the map again
	users.Scan(func(key, value string) bool {
		fmt.Printf("%s %s\n", key, value)
		return true
	})
	fmt.Printf("\n")

	// Output:
	// user:1 Jane
	// user:2 Andy
	// user:3 Steve
	// user:4 Andrea
	// user:5 Janet
	// user:6 Andy
	//
	// user:2 Andy
	// user:3 Steve
	// user:4 Andrea
	// user:6 Andy
}
```

Compiled it with `tinygo build -target wasi -o test.wasm`

aykevl, 2024-10-25:
Unfortunately I still cannot reproduce this issue. This is with `tinygo version 0.33.0 linux/arm64 (using go version go1.22.2 and LLVM version 18.1.2)` so very similar to yours.

Can you tell me:
- which OS are you using?
- how did you get `tinygo`? (Distribution package, manual build, or our own binaries?)

lasarocamargos, 2024-10-25:
Sure. I am using ubuntu 24.04.1 LTS, on a multipass VM in OSX.
`Linux dev-vmc 6.8.0-47-generic #47-Ubuntu SMP PREEMPT_DYNAMIC Fri Sep 27 22:03:50 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux`

I manually built tinygo.

aykevl, 2025-10-01:
Still can't reproduce this error, with `tinygo version 0.40.0-dev-109e0767 linux/arm64 (using go version go1.24.0 and LLVM version 20.1.8)`.

@lasarocamargos do you still have this problem? Please comment if you do, I'm going to close this for now since I haven't heard anybody else run into this (and the problem may have gone away).

closed 2025-10-01 by aykevl