tinygo-org/tinygo #4551: darwin releases missing wasm-opt?

dkegel-fastly, 2024-10-26:
Shades of https://github.com/tinygo-org/tinygo/issues/3977, but for darwin:

```
% tar -tf ~/Downloads/tinygo0.33.0.darwin-arm64.tar.gz | grep -i wasm-opt || echo bupkis
bupkis
% tar -tf ~/Downloads/tinygo0.33.0.darwin-amd64.tar.gz | grep -i wasm-opt || echo bupkis 
bupkis
% tar -tf ~/Downloads/tinygo0.33.0.linux-amd64.tar.gz | grep -i wasm-opt || echo bupkis 
tinygo/bin/wasm-opt
```

Persists in 0.34-pre

deadprogram, 2024-10-26:
The homebrew recipe installs it https://github.com/tinygo-org/homebrew-tools/blob/master/tinygo.rb#L6

dkegel-fastly, 2024-10-28:
Right.  I guess that's how most people get it.

I use brew for everything but go and tinygo, I don't trust it to not screw up...

deadprogram, 2024-12-11:
Closing since it is not the typical use case.

closed 2024-12-11 by deadprogram