tinygo-org/tinygo #4557: Cannot build anything with tinygo on windows

RodargDev, 2024-10-27:
I have followed all steps given in this installation guide (https://tinygo.org/getting-started/install/windows/)

I currently have go and tinygo installed through scoop with versions:
```go version go1.23.2 windows/amd64```
```tinygo version 0.34.0 windows/amd64 (using go version go1.23.2 and LLVM version 18.1.2)```

I have a small program with the following code:
```
package main

import "fmt"

func main() {
	fmt.Println("Hello, World!")
}
```

But when i try to use tinygo to build this for the pico (Or any other targer) i get a huge stacktrace with a lot of gibberish characters. Only the start and end of the stacktrace give some context into whats going on.

The command i use
```tinygo flash -target=pico .\main.go```

The output i get in console
```
PS C:\Projects\WaterLevelSensor> tinygo flash -target=pico .\main.go
error: unable to make temporary file: No such file or directory
# 1 "C:\\Users\\info\\scoop\\apps\\tinygo\\current\\lib\\compiler-rt-builtins\\arm\\aeabi_cdcmp.S"
error: unable to make temporary file: No such file or directory
# 1 "<builerror: unable to make temporary file: No such file or directory
t-in>" 1
# 1 "<built-in>" 3
# 1 "C:\\Users\\info\\scoop\\apps\\tinygo\\current\\lib\\compiler-rt-builtins\\arm\\aeabi_cfcmp# 1# 377 "<built-in>" 3
# 1 "<command line>" 1
 "C:\.S"
\Users\\info\\scoop\\apps\\tinygo\\curren# error: unable to make temporary file: No such file or directory
t\\# 1 "<built-in>" 1
lib\\compile# 1 "<built-in>" 31 "<buir
lt-in>" 2
-# 1 rt-built"C:\\User# 377 "<built-in>" 3
# s\\# 1 "C:\\Users\\info\\scoop\\apps\\tininfo\\scygo\\cins\\arm\\aourrent\\lib\\compieabi_dcmp.S"
o# 1 1 ler-rp"<bt-builtiuierror: unable to make temporary file: No such file or directoryns\\arl"<com
\\ammt-and line>" 1
i\\aeabi_#n>" 1
# ppfc1 "<s\\tinygo\\current\\lib\\com 1 "<bmbpiler-uilt-in>"p.S"
uilt-in>" 3
 2
# 1# 1 "C:\\Users\\infrt-builtins\\ar##o\\scoop\\apps\   "<1 "C:\\Users\\info\\scoop\\apps\\tinygo\\current\\l\t377 "ib\built-in>" 1
<built-inygo\\current\i\com\lib\\com# 1 piln>" 3piler-er-m\\aeabi_cdcmp.S"rt-r" 2
built

t-ins\builtins\\arm# \\<#error: error: 1error: unable to make temporary file: No such file or directory
arm\unable to make temporary file: No such file or directory
\unable to make temporary file: No such file or directory
 "<co 1 "C:\\Usersaeabi_cfcmbuilt-in>" 3
m#p\aeabi_id # 377.S" 2
1 "C:\#ivmodmand line>"# 1 "C:\\User 1

# Right here a lot of characters
 
.global literals
literals:
.ltorg

.end
# 17 "C:\\Users\\info\\scoop\\apps\\tinygo\\current\\targets\\pico-boot-stage2.S" 2

failed to build C:\Users\info\scoop\apps\tinygo\current\src\device\arm\cortexm.S: exit status 1
```

Ive tried resolving it and have been looking around but it appears i cant find anyone with the same problem, it looks like its somewhat permission based but when running this through admin terminal it still wont work.

aykevl, 2024-10-29:
My first guess would be maybe something related to this PR? https://github.com/tinygo-org/tinygo/pull/4513

One thing that would be useful to know is if you can create an empty file `empty.S` (name doesn't matter) and then run:

    tinygo clang -c -o empty.o empty.S

This _should_ work, though I expect it will fail in your case. That would be an interesting datapoint.

soypat, 2024-11-05:
@aykevl Atendee ran into this at Gophercon and your command worked on their PC. They are still getting the above error.

soypat, 2024-11-05:
@aykevl with @pieterlouw we found a fix! 

```
set TEMP=C:/a/path/that/exists/and/is/writeable
tinygo flash -target=pico yourprogram
```

aykevl, 2024-11-07:
Ah yes, if the temporary directory is not writable all kinds of issues will turn up.

I don't think there's much we can do on the TinyGo side. I guess we could check whether `goenv.Get("TEMP")` is writable, but that's about it.

colececil, 2024-11-11:
> @aykevl with @pieterlouw we found a fix!
> 
> ```
> set TEMP=C:/a/path/that/exists/and/is/writeable
> tinygo flash -target=pico yourprogram
> ```

I'm running into the same issue, but this workaround doesn't seem to be doing anything for me. Is there any way to tell where it's trying and failing to write the file?

My versions of everything are almost the same as @RodargDev's:
> tinygo version 0.34.0 windows/amd64 (using go version go1.22.3 and LLVM version 18.1.2)

sago35, 2024-11-11:
There is write permission for `TEMP`, but the issue still occurs.
Changing `TEMP` to another location results in the same outcome.

The issue was reproducible in my environment as well, so I'll investigate further.

aykevl, 2024-11-11:
If you are able to build TinyGo, it would be very useful if you could try to comment out this line:

https://github.com/tinygo-org/tinygo/blob/2a76ceb7dd5ea5a834ec470b724882564d9681b3/builder/tools.go#L34-L37

jeffcaiz, 2024-11-11:
Save issue here, but `mkdir C:\Temp`, create the temp dir make things work now.

I use API Monitor to watch and notice this. 

But no idea what part of code this dir is introduced.

colececil, 2024-11-12:
> Save issue here, but `mkdir C:\Temp`, create the temp dir make things work now.
> 
> I use API Monitor to watch and notice this.
> 
> But no idea what part of code this dir is introduced.

I just tried this and can confirm that it works! 😲 Once I created `C:\Temp`, TinyGo was able to build my project! Very weird - I guess that directory name must be hard coded somewhere? It's definitely not what my `TEMP` environment variable was set to.

aykevl, 2024-11-12:
Here is a patch that should hopefully fix this problem: https://github.com/tinygo-org/tinygo/pull/4596

deadprogram, 2024-12-06:
Tagging to close this issue on next release.

closed 2024-12-21 by deadprogram