tinygo-org/tinygo #4568: `select` dead locked by `time.After`

d-enk, 2024-10-30:
on my machine
tinygo version 0.34.0 linux/amd64 (using go version go1.23.2 and LLVM version 18.1.2

[go.dev/play](https://go.dev/play/p/uryoB6fAI3v)

```go
package main

import (
	"fmt"
	"time"
)

func debug(v ...any) {
	fmt.Println(time.Now().Format(".000000"), v)
}

func Run(input <-chan int) {
	taskDone := make(chan struct{}, 1)

	for {

		debug("SELECT")

		select {
		case <-time.After(time.Millisecond):
			panic("timeout")

		case <-taskDone:
			debug("WAKEUP BY TASK DONE")

		case value := <-input:
			debug("WAKEUP BY INPUT")

			go func() {
				fmt.Println("PROCESS...", value)

				select {
				case taskDone <- struct{}{}:
				default: // already notified
				}
			}()

		}

		debug("SELECT DONE")
	}
}

func main() {
	input := make(chan int, 3)

	for i := 0; i < cap(input); i++ {
		input <- i
	}
	// close(input)

	Run(input)
}

```

output by [playgound](https://play.tinygo.org/)
```
​.001075 [SELECT]
​.003140 [WAKEUP BY INPUT]
​.003225 [SELECT DONE]
​.003240 [SELECT]
​.003265 [WAKEUP BY INPUT]
​.005485 [SELECT DONE]
​.005555 [SELECT]
​.005575 [WAKEUP BY INPUT]
​.005700 [SELECT DONE]
​.005725 [SELECT]
​.005845 [PROCESS... 0]
​.006030 [PROCESS... 1]
​.006050 [PROCESS... 2]
​.006075 [WAKEUP BY TASK DONE]
​.006085 [SELECT DONE]
​.006095 [SELECT]
​.006105 [WAKEUP BY TASK DONE]
​.006120 [SELECT DONE]
​.006130 [SELECT]
```

but then ​`panic: timeout` expected

aykevl, 2024-10-30:
I can confirm this bug. Also verified it's present in v0.33.0, so it's not a regression in the latest release.

aykevl, 2024-10-30:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/4569

closed 2024-12-21 by deadprogram