tinygo-org/tinygo #4588: Segmentation fault when compiling Wasm module in tests

inliquid, 2024-11-07:
We have a segmentation violation errors in some tests during `tinygo build <...>` invocation. They start failing very recently after [this change](https://github.com/knqyf263/go-plugin/pull/65/commits/7020e361f47da7e3567f45663016d8e743d7c0bd) (this is not necessarily connected) and not in every run. 

Full error message can be seen by following below links, message in both cases more or less similar:
```
<...>
tinygo build -o tests/well-known/plugin/plugin.wasm -scheduler=none --no-debug --target=wasi tests/well-known/plugin/plugin.go
SIGSEGV: segmentation violation
PC=0x81301c m=0 sigcode=18446744073709551610 addr=0x3e900000ed0
<...>
```

This error happens with different versions of `tinygo`. For example:
1. tinygo-v0.29.0 (go1.20.14)
    - [Passed](https://github.com/knqyf263/go-plugin/actions/runs/10559633234/job/29252147334)
    - [Failed](https://github.com/knqyf263/go-plugin/actions/runs/10570578373/job/29285270109)
2. tinygo-v0.31.2 (go1.22.8)
    - [Passed](https://github.com/knqyf263/go-plugin/actions/runs/11725190050/job/32660864555)
    - [Failed](https://github.com/knqyf263/go-plugin/actions/runs/11729232028/job/32674393261)
  

This is code which `tinygo` failed to compile -> [plugin.go](https://github.com/knqyf263/go-plugin/blob/main/tests/well-known/plugin/plugin.go).

I can't reproduce this locally. Tried testing with Ubuntu 22.04 and older Ubuntu 20.04 with exact same Go and TinyGo versions as in failing tests, but everything worked well. 

So this seems to be some sort of flaky test, probably environment-dependent (github runner). 

To reproduce you need to clone https://github.com/knqyf263/go-plugin and run `make test`.

aykevl, 2024-11-08:
If there's no way to reproduce locally, it's going to be very difficult to debug this. Especially because this appears to be a memory corruption issue.

Have you tried simply running the test many times, possibly with `tinygo clean` in between runs?

inliquid, 2024-11-08:
> If there's no way to reproduce locally, it's going to be very difficult to debug this. Especially because this appears to be a memory corruption issue.

Yes, I understand, maybe anyone else have encountered same error and can add more context...

> Have you tried simply running the test many times, possibly with tinygo clean in between runs?

Thanks, I'll try to run test with this command in a loop and get back with result, whole test takes couple of minutes, so unfortunately it's not easy to produce a lot of attempts.