tinygo-org/tinygo #4597: nRF52: update from 0.33 -> 0.34 causes 500x idle-current

perttierkkila, 2024-11-13:
For ultra-low power applications where all peripherials are disabled and usually under 1.5uA SystemON currents, update to 0.34 was a bad mistake which caused 600uA idle current. After debugging, something keeps requesting HF-clock.

I tried to check diffs between 0.33 and 0.34 and I can't find any changes related to nRF52-parts. The problem is caused under [initAll()](https://github.com/tinygo-org/tinygo/blob/release/src/runtime/runtime.go#L13)-code. I tried to dig that part also but no luck as it goes to compiler-part.

Example code:
`const led = machine.P0_20

func main() {
	arm.EnableIRQ(uint32(nrf.IRQ_RTC1)) // workaround to keep IRQs working, already fixed in dev

	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Set(nrf.CLOCK.HFCLKSTAT.Get()>>16 == 1)
		time.Sleep(10 * time.Millisecond)
		led.Low()
		time.Sleep(500 * time.Millisecond)
	}
}`

Serial, GC and scheduler disabled, flashing with command:
`tinygo flash -size=short -scheduler=none -gc=none -serial=none -target=pca10040 -programmer command main.go`

Output:
- 0.34: led blinks, current 600uA+ (led uses minimal current compared to HF-clock)
- 0.33: no blinking, current under 1.5uA

Also tried to play with -opt values, no affect. Expect -opt=0 which keeps current under 1.5uA but code is not running either. Any ideas which causes this issue?

sago35, 2024-12-07:
@aykevl Please give some hints.

aykevl, 2024-12-14:
How do I reproduce this issue?
It makes sense that HFCLK is running while the code is running. So I think when reading from this register, it will always be set to "1".

I have the following test code, flashed on a PCA10056:

```go
package main

import (
	"device/nrf"
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Set(nrf.CLOCK.HFCLKSTAT.Get()>>16 == 1)
		time.Sleep(500 * time.Millisecond)
		led.Low()
		time.Sleep(500 * time.Millisecond)
	}
}
```

It blinks in both v0.33.0 and v0.34.0.

aykevl, 2024-12-14:
> After debugging, something keeps requesting HF-clock.

What do you mean by this? After running `tinygo gdb`?
Yes, the chip will continue to use lots of current after doing on-chip debugging. The only way to get to normal (low current) behavior is by power cycling. This is a hardware limitation, not a TinyGo bug.

perttierkkila, 2025-01-07:
You must use option -serial=none to test actual consumption. Default setting with UART always requires HF-clock and hides a problem.

I just tried to find issues with gdb, no luck. Actual tests were done in standard-mode with option -serial=none

perttierkkila, 2025-01-21:
Tested with 0.35, behaves like 0.34 => 600uA+ sleep-current.

I assume there is now something running extensively in background which causes CPU wakeups and this requests HF-clock.

Measurement from 0.35:
[image]

perttierkkila, 2025-01-21:
And same measurement, ONLY changed thing is tinygo version 0.35->0.33. Same code, same hardware and same command.

[image]

deadprogram, 2026-10-10:
This should be fixed in 0.43.0. In 0.34 the runtime started seeding fastrand from the hardware RNG at startup (f3dfe1d4), but the nRF RNG was never stopped after the read, so it kept running. 235b14b3 now stops the RNG after each read.

Closing, but please reopen if you still see the higher idle current.

closed 2026-10-10 by deadprogram