tinygo-org/tinygo #4598: Missing various net/http packages in std

inliquid, 2024-11-13:
This is follow up of #4580 as the initial issue was closed.

I have same issue when trying to build module, which imports quite complex `*.proto` definitions:
```
$ tinygo build -o wasm/plugin.wasm -scheduler=none -target=wasip1 --no-debug ./wasm
../../../../pkg/mod/golang.org/x/net@v0.30.0/http2/transport.go:26:2: package net/http/httptrace is not in std (/home/***/.cache/tinygo/goroot-8cbcb1d9555609a60075264735fce03217a624835ebf583eb96aa0c5bf08e840/src/net/http/httptrace)
../../../../pkg/mod/google.golang.org/grpc@v1.66.0-dev/internal/transport/proxy.go:29:2: package net/http/httputil is not in std (/home/***/.cache/tinygo/goroot-8cbcb1d9555609a60075264735fce03217a624835ebf583eb96aa0c5bf08e840/src/net/http/httputil)
```

I would like to be able to compile these dependencies into my Wasm modules. And I see a lot of progress, previously there were much more scaring compiler errors, now when `reflect` support is there, only those left. At the moment I have to regenerate these `*.proto` definitions using alternative generators.

scottfeldman, 2025-01-18:
Is this solved by tinygo-org/net#34?

inliquid, 2025-01-18:
Hi @scottfeldman, I don't see `httptrace` or `httputil` was added in that PR.

scottfeldman, 2025-01-19:
> Hi [@scottfeldman](https://github.com/scottfeldman), I don't see `httptrace` or `httputil` was added in that PR.

Oh, you're right.  My bad.  

Can you supply your program or a test program I can use to repro?  Then I can look into adding httptrace and httputil.  THanks.

deadprogram, 2025-08-10:
`httptrace` was just added in https://github.com/tinygo-org/net/pull/39

davecheney, 2026-10-24:
`net/http/httptrace` and `net/http/httputil` are now included in TinyGo’s pinned net dependency. See https://github.com/tinygo-org/net/pull/43 and https://github.com/tinygo-org/tinygo/pull/5298.

These standalone programs check the reported missing packages without making network requests.

`httptrace-check.go`:
```go
package main

import (
    "context"
    "fmt"
    "net/http/httptrace"
)

func main() {
    trace := &httptrace.ClientTrace{}
    ctx := httptrace.WithClientTrace(context.Background(), trace)
    if httptrace.ContextClientTrace(ctx) != trace {
        panic("client trace was not preserved")
    }
    fmt.Println("httptrace: OK")
}
```

`httputil-check.go`:
```go
package main

import (
    "fmt"
    "net/http"
    "net/http/httputil"
    "strings"
)

func main() {
    req, err := http.NewRequest("GET", "http://example.test/check", nil)
    if err != nil {
        panic(err)
    }
    dump, err := httputil.DumpRequest(req, false)
    if err != nil {
        panic(err)
    }
    if !strings.HasPrefix(string(dump), "GET /check HTTP/1.1\r\n") {
        panic("unexpected request dump: " + string(dump))
    }
    fmt.Println("httputil: OK")
}
```

Validated with TinyGo 0.44.0-dev-eae3f875 on darwin/arm64 using Go 1.27.3 and LLVM 23.1.4:
```sh
tinygo run httptrace-check.go
tinygo run httputil-check.go
tinygo build -target=wasm -o httptrace-check.wasm httptrace-check.go
tinygo build -target=wasip1 -o httptrace-check-wasip1.wasm httptrace-check.go
tinygo build -target=wasm -o httputil-check.wasm httputil-check.go
tinygo build -target=wasip1 -o httputil-check-wasip1.wasm httputil-check.go
```
Both native runs printed their respective `OK` messages, and all four WebAssembly builds succeeded. The WebAssembly binaries were not run.

Closing the missing-package report as completed. This does not establish compatibility with every gRPC, protobuf, or Kubernetes dependency. Please open a separate issue with a reproducer for any remaining compilation errors.

closed 2026-10-24 by davecheney