tinygo-org/tinygo #4605: type net.Interface has no field or method Addrs

deepch, 2024-11-16:
`
ifaces, _ := net.Interfaces()
	for _, i := range ifaces {
		addrs, err := i.Addrs()
`

internal/util/util.go:86:19: i.Addrs undefined (type net.Interface has no field or method Addrs)

scottfeldman, 2025-01-18:
Hi, the TinyGo 'net' package has limited support for Interfaces, as you've discovered.  We can add Addrs() to interface.go, but it would just be a stub to satisfy the complier.  Do you need the interface addresses, or are you just trying to get something to compile and bumped into this?

davecheney, 2026-10-24:
The missing `net.Interface.Addrs` method is now present. This program compiles for both `wasm` and `wasip1`:

```go
package main

import "net"

func main() {
    ifaces, err := net.Interfaces()
    if err != nil {
        panic(err)
    }
    for _, iface := range ifaces {
        if _, err := iface.Addrs(); err != nil {
            panic(err)
        }
    }
}
```

```sh
tinygo build -target=wasm -o interface-addrs-check.wasm interface-addrs-check.go
tinygo build -target=wasip1 -o interface-addrs-check-wasip1.wasm interface-addrs-check.go
```

Checked with TinyGo 0.44.0-dev-eae3f875 on darwin/arm64 using Go 1.27.3 and LLVM 23.1.4. The validated program also prints a success message after the loop; the version above omits that output.

There is still a runtime limitation. Native execution on macOS panics at `net.Interfaces()` with `Interfaces not implemented`, before reaching `Addrs`. The browser and WASI implementations return an empty interface list, so this loop does not exercise `Addrs` there. The WebAssembly binaries were compiled but not run.

The original missing-method compilation error is resolved, but working interface enumeration is not established. Leaving this issue open.