tinygo-org/tinygo #4609: How run tests?

eleimt, 2024-11-17:
Hello!
I tried run simple test
```
func TestSimple(t *testing.T) {
	t.Errorf("error")
}
```
but get error
```
$ tinygo test
FAIL    myproject/cmd/myproject     0.000s
# myproject/internal/config
..\..\internal\config\config.go:34:22: undefined: machine.UART0_TX_PIN
..\..\internal\config\config.go:35:22: undefined: machine.UART0_RX_PIN
..\..\internal\config\config.go:40:22: undefined: machine.UART0_TX_PIN
..\..\internal\config\config.go:41:22: undefined: machine.UART0_RX_PIN
```
I have `config.go` file
```
func MustLoad() *Config {
	return &Config{
		Receiver: Uart{
			Uart:     machine.UART0,
			Tx:       machine.UART0_TX_PIN,
			Rx:       machine.UART0_RX_PIN,
			BaudRate: 115200,
		},
		// ...
	}
}
```
Help me please...

aykevl, 2024-11-18:
You are leaving out a lot of code. What's this `Config` struct? What's this `Uart` struct?

Note that you probably don't want to use UART on the pico, most people use the default serial output over USB (which is USB-CDC, not UART), and is available from `machine.Serial`.

eleimt, 2024-11-18:
@aykevl It's just structure, example Led: machine.LED etc. For run tests I using USB. Without mapping Pins tests works normal.

eleimt, 2024-12-06:
@aykevl

deadprogram, 2024-12-12:
Hello @eleimt could you please provide a complete code sample that shows your specific issue? Thank you.

eleimt, 2024-12-18:
> Hello @eleimt could you please provide a complete code sample that shows your specific issue? Thank you.

File 1: `cmd/abc/main.go`
```go
package main

import (
	"machine"
	"time"
)

func main() {

	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 500)

		led.High()
		time.Sleep(time.Millisecond * 500)
	}
}

```

File 2 `cmd/abc/main_test.go`
```go
package main

import (
	"testing"
)

func TestBuild(t *testing.T) {
	t.Errorf("Should not produce an error")
}
```

Run command `tinygo test ./cmd/abc` result:
```
FAIL    car/cmd/abc     0.000s
# car/cmd/car
cmd\abc\main.go:10:17: undefined: machine.LED
```

I'm expecting
```
--- FAIL: TestBuild (0.00s)
    Should not produce an error
FAIL
FAIL    abc/cmd/abc     0.042s
```

eleimt, 2025-03-05:
@deadprogram Do you need any more information?

eleimt, 2025-04-15:
@aykevl @deadprogram Help, please :)

aykevl, 2025-04-16:
Ok, in that case you would need to use `tinygo test -target=pick ./cmd/abc`. But I don't think we really support this. It's difficult to run such tests on real hardware.

eleimt, 2025-04-16:
> ~Ok, in that case you would need to use `tinygo test -target=pick ./cmd/abc`. But I don't think we really support this. It's difficult to run such tests on real hardware.~
> 
> Unfortunately we don't support running Go tests on real hardware at the moment.

Okay. How I can run tests on the host system?

aykevl, 2025-04-17:
You can use `go test` or `tinygo test` as usual. Though you will likely have to modify your program to not import the machine package in tests.

closed 2025-04-17 by eleimt