tinygo-org/tinygo #4624: build error: net/smtp module reports undefined: tls.Conn

everyx, 2024-11-21:
```console
❯ tinygo run verifier.go 
# net/smtp
/usr/lib/go/src/net/smtp/smtp.go:72:24: undefined: tls.Conn
/usr/lib/go/src/net/smtp/smtp.go:172:25: undefined: tls.Conn
```

<details>
<summary>verifier.go</summary>

```go
package main

import (
	"encoding/json"
	"fmt"

	emailVerifier "github.com/AfterShip/email-verifier"
)

type Result struct {
	Data  *emailVerifier.Result `json:"data"`
	Error error                 `json:"error"`
}

func (p Result) ToJson(result Result) (*string, error) {
	bytes, err := json.Marshal(result)

	if err != nil {
		fmt.Println("Error converting to JSON:", err)
		return nil, err
	}

	json := string(bytes)

	return &json, nil
}

//export add
func verify(email string) string {
	var result Result

	verifier := emailVerifier.NewVerifier().EnableAutoUpdateDisposable()

	ret, err := verifier.Verify(email)
	if err != nil {
		result = Result{
			Data:  nil,
			Error: err,
		}
	} else {
		result = Result{
			Data:  ret,
			Error: nil,
		}
	}

	bytes, err := json.Marshal(result)
	if err != nil {
		fmt.Println("Error converting to JSON:", err)
		return fmt.Sprintf("{error:%s}", err.Error())
	}

	return string(bytes)
}

func main() {
	fmt.Println(verify("test@example.com"))
}
```

</details>

scottfeldman, 2025-01-18:
This isn't working because current TinyGo TLS support is only for baremetal devices with mbed firmware for offloading the TLS protocol.  TinyGo will redirect tls calls down to mbed firmware.

Seems we need someway at compile time to pick which 'net' and 'crypto/tls' packages for the target.  If targeting WASM or non-baremetal, pick the standard Go packages.  If targeting baremetal, pick the TinyGo 'net' and 'crypto/tls' packages.

deadprogram, 2026-10-10:
This is fixed. `tls.Conn` was added in 0.42.0 by #5490, so `net/smtp` now builds.

Closing, but please reopen if you still see it.

closed 2026-10-10 by deadprogram