tinygo-org/tinygo #4630: the network does not work without drivers? Lookup of host name 'www.com' failed: Netdev not set

deepch, 2024-11-26:
Mac OS M2

```
	telnet, err := net.Dial("tcp4", fmt.Sprintf("www.com:%d", 80))
	if err != nil {
		log.Fatalln(err)
	}

```
`Lookup of host name 'www.com' failed: Netdev not set`

`tinygo run cmd/client/main.go`

davecheney, 2026-10-26:
This still does not work correctly on macOS. The self-contained sample below compiles with TinyGo but fails before an HTTP exchange on darwin/arm64:

```text
listen: Lookup of host name '127.0.0.1' failed: Netdev not set
```

It starts a loopback HTTP server and checks a request from an HTTP client, without relying on DNS or an external service.

```sh
tinygo run main.go
```

```go
package main

import (
	"fmt"
	"io"
	"net"
	"net/http"
	"os"
	"time"
)

func main() {
	if err := check(); err != nil {
		fmt.Fprintln(os.Stderr, err)
		os.Exit(1)
	}
}

func check() error {
	listener, err := net.Listen("tcp", "127.0.0.1:0")
	if err != nil {
		return fmt.Errorf("listen: %w", err)
	}
	defer listener.Close()

	server := &http.Server{
		Handler: http.HandlerFunc(func(w http.ResponseWriter, r *http.Request) {
			if _, err := io.WriteString(w, "hello from TinyGo\n"); err != nil {
				fmt.Fprintln(os.Stderr, "write response:", err)
			}
		}),
	}
	defer server.Close()

	done := make(chan error, 1)
	go func() {
		done <- server.Serve(listener)
	}()

	client := &http.Client{Timeout: 5 * time.Second}
	defer client.CloseIdleConnections()
	response, err := client.Get("http://" + listener.Addr().String() + "/")
	if err != nil {
		return fmt.Errorf("GET: %w", err)
	}
	defer response.Body.Close()

	body, err := io.ReadAll(response.Body)
	if err != nil {
		return fmt.Errorf("read response: %w", err)
	}
	if response.StatusCode != http.StatusOK || string(body) != "hello from TinyGo\n" {
		return fmt.Errorf("unexpected response: status=%d body=%q", response.StatusCode, body)
	}
	if err := server.Close(); err != nil {
		return fmt.Errorf("close server: %w", err)
	}
	if err := <-done; err != http.ErrServerClosed {
		return fmt.Errorf("Serve returned %v, want http.ErrServerClosed", err)
	}
	fmt.Print("PASS: HTTP client and server exchanged ", string(body))
	return nil
}
```

riobard, 2026-10-26:
@davecheney On Linux it kinda works but in a dangerous way: `http.Client` sends plain HTTP cleartext to "https://" url without doing any TLS handshakes… Had to change API keys after figuring out what's going wrong.