tinygo-org/tinygo #4659: 'time' package adds 'syscall/js' dependencies to wasip1 build

falconandy, 2024-12-15:
Without `time` package:

```go
package main

//go:wasmexport add
func add(a, b int32) int32 {
	//_ = time.Time{}
	return a + b
}

func main() {}
```

`GOOS=wasip1 GOARCH=wasm tinygo build -o ./wasmtest.wasm -buildmode=c-shared --no-debug -scheduler=none .`

Result size: 9563 bytes

[image]

With `time` package:
```go
package main

import (
	"time"
)

//go:wasmexport add
func add(a, b int32) int32 {
	_ = time.Time{}
	return a + b
}

func main() {}

```

`GOOS=wasip1 GOARCH=wasm tinygo build -o ./wasmtest.wasm -buildmode=c-shared --no-debug -scheduler=none .`

Result size: 13698 bytes

[image]

Is it intended to have `syscall/js` dependencies in wasip1 builds?

mattjohnsonpint, 2024-12-23:
I can confirm this happens when relying on `GOOS=wasip1 GOARCH=wasm`, but it does not happen when passing `-target=wasip1`.  Same as in #4439.  (It's not specific to the `time` package.)

Occurs both on 0.34.0 and 0.35.0.

deadprogram, 2024-12-24:
Please see https://github.com/tinygo-org/tinygo/pull/4437 and https://github.com/tinygo-org/tinygo/pull/4447 for details.

deadprogram, 2025-03-04:
Closed as part of v0.36.0

closed 2025-03-04 by deadprogram