tinygo-org/tinygo #4670: wasm-unknown: cannot build binaries without main

orsinium, 2024-12-20:
I have a project using the following `target.json`:

```json
{
  "llvm-target": "wasm32-unknown-unknown",
  "cpu": "generic",
  "features": "+mutable-globals,+nontrapping-fptoint,+sign-ext,+bulk-memory",
  "build-tags": ["tinygo.wasm", "wasm_unknown"],
  "goos": "linux",
  "goarch": "arm",
  "linker": "wasm-ld",
  "rtlib": "compiler-rt",
  "scheduler": "none",
  "gc": "leaking",
  "cflags": ["-mno-bulk-memory", "-mnontrapping-fptoint", "-msign-ext"],
  "ldflags": [
    "--allow-undefined",
    "--no-demangle",
    "--stack-first",
    "--no-entry",
    "-zstack-size=14752"
  ],
  "extra-files": ["src/runtime/asm_tinygowasm.S"]
}
```

In tinygo 0.33.0, the presence of `"--no-entry"` linker flag allows the project to be build without defining `main` function. However, doing so in 0.34.0 adds `env.main/main(i32)` import in the wasm binary.  Bisecting showed that the issue is introduced in #4451.

orsinium, 2024-12-22:
Fixed by adding `-buildmode c-shared` in 0.35.0 ✅  Found solution here: https://github.com/dylibso/xtp-go-bindgen/pull/30

closed 2024-12-22 by orsinium

deadprogram, 2024-12-22:
It would be good to add this info someplace in the TinyGo docs for WASM, probably.