tinygo-org/tinygo #4673: rp2040 / rp2380 xoscFreq setting is nonfuctional

cibomahto, 2024-12-30:
The RP2040 and 2350 board files all specify a parameter for the external crystal frequency, [xoscFreq](https://github.com/tinygo-org/tinygo/blob/release/src/machine/board_pico.go#L38). This appears to be so that a different external crystal could be used on a particular board.

Unfortunately, the [PLL configurations for the system and USB clock](https://github.com/tinygo-org/tinygo/blob/52983794d702af7a00833ae12b0d2e7175e46017/src/machine/machine_rp2_clocks.go#L149) are hard-coded to assume a 12MHz input clock, so this setting doesn't actually work. Finding an optimal setting for the PLL [is complicated](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_clocks/scripts/vcocalc.py), and it's probably best not to do this on the fly during the startup code.

I can think of two straightforward solutions:
1. Remove the xoscFreq setting. The majority of RP2040 or RP2350 designs should use this frequency.
2. Add the two PLL configurations to the board (or target) configuration, so that a custom crystal frequency can be used correctly.

I think option 1 better, unless  someone has a board with a nonstandard clock that they need to support. I couldn't find examples of other machines/boards that support a similar feature. If either of these sound reasonable, I'd be happy to make a pull request.

cc @soypat @deadprogram

cibomahto, 2025-01-06:
I created a [clock configuration structure](https://github.com/tinygo-org/tinygo/pull/4674/commits/586c88e4cdbaf4fbad178f3ecca171b53ba98a6b#diff-4babe1e3aead6c34db2f8de3a7d1680b3f073b142370939eebf3d9397f5666a2), made a default in [machine_rp2_2xxx.go](https://github.com/tinygo-org/tinygo/pull/4674/commits/586c88e4cdbaf4fbad178f3ecca171b53ba98a6b#diff-f4c70a0146ba047fe88ad88a2bd140a6d3001ef7aa845b7b69fb27931ada2532), then modified the [board files](https://github.com/tinygo-org/tinygo/pull/4674/commits/586c88e4cdbaf4fbad178f3ecca171b53ba98a6b#diff-c40f5ef7da9ed2e0faab5e546d8ca6b875a50f55af44223617dd9d9e0b87dd0f) to create a new object based on the default.

This appears to work if I build a target locally, however the [CI test fails](https://app.circleci.com/pipelines/github/tinygo-org/tinygo/12161/workflows/65a2f17e-4fa6-4f9a-917a-676f3cf63cd1/jobs/45712?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-checks-link&utm_content=summary&invite=true#step-118-7606_89) because it appears to be building a test for WASM based on the board file, but not including the machine backend. So I imagine that this wasn't the right way to do this :-)

If anyone has an example of a TinyGo way to achieve this let me know. Otherwise I'm out of guesses.

deadprogram, 2025-02-04:
Looking into this today...

cibomahto, 2025-02-04:
@deadprogram thanks! See https://github.com/tinygo-org/tinygo/pull/4674 for some more discussion about this

soypat, 2025-03-12:
I believe this was addressed in https://github.com/tinygo-org/tinygo/pull/4768

xoscfreq is now functional and CPU frequency is calculated at compile time, huge wins! RP2040 is now running at 200MHz in dev :)

deadprogram, 2025-03-19:
Now closing with 0.37.0 release. Thanks!

closed 2025-03-19 by deadprogram