tinygo-org/tinygo #4682: `GOOS=wasip1 GOARCH=wasm` not working, `gojs::runtime.ticks` error

KieranP, 2025-01-04:
According to https://tinygo.org/docs/guides/webassembly/wasi/, I can use `GOOS=wasip1 GOARCH=wasm` just like GoLang supports. But it doesn't appear to work.

```
$ GOOS=wasip1 GOARCH=wasm tinygo build -o output.wasm
$ wasmtime run output.wasm
Error: failed to run main module `output.wasm`

Caused by:
    0: failed to instantiate "output.wasm"
    1: unknown import: `gojs::runtime.ticks` has not been defined
```

Using -target works:

```
$ tinygo build -o output.wasm -target=wasip1
$ wasmtime run output.wasm
  [runs app as expected]
```

Has support for GOOS/GOARCH been removed?

deadprogram, 2025-01-04:
I think this is duplicate of https://github.com/tinygo-org/tinygo/issues/4659

deadprogram, 2025-02-21:
I just created #4746 to try to address this issue.

deadprogram, 2025-03-04:
Closed as part of v0.36.0

closed 2025-03-04 by deadprogram