tinygo-org/tinygo #4687: `tinygo build -o /path/to/out` does not create intermediate directories, like `go build` does

jphastings, 2025-01-05:
- `go build -o /path/to/out .` creates `/path/to/` if needed, before starting the build.
- `tinygo build -o /path/to/out .` **does not** create /path/to/` if the directory doesn't exist, failing the build with the slightly confusing message "open /path/to/out: no such file or directory"

This is confusing behaviour, but also makes the use of build tooling, like [GoReleaser](https://goreleaser.com/), challenging as substituting `tinygo` in for `go` in the build configuration doesn't work as expected, and throws the error: "open [your output path]: no such file or directory".

<details>
<summary>Repoducing the issue</summary>

Get into a Go project directory that can compile with Go and TinyGo.

```sh
# Create a space for the experiment & show the structure
$ mkdir -p /tmp/experiment

$ tree /tmp/experiment
/tmp/experiment

# Build to a directory that does not exist with tinygo: failure
$ tinygo build -o /tmp/experiment/somedir/tinygo-built .
open /tmp/experiment/somedir/tinygo-built: no such file or directory

$ tree /tmp/experiment
/tmp/experiment

# Create the directory manually, then try again: success
$ mkdir /tmp/experiment/somedir
$ tinygo build -o /tmp/experiment/somedir/tinygo-built .

$ tree /tmp/experiment
/tmp/experiment
└── somedir
    └── tinygo-built

# Remove that directory, then build with go instead: success
$ rm -rf /tmp/experiment/somedir
$ tree /tmp/experiment
/tmp/experiment

$ go build -o /tmp/experiment/somedir/go-built .

$ tree /tmp/experiment
/tmp/experiment
└── somedir
    └── go-built

```

</details>

jphastings, 2025-01-05:
For GoReleaser specifically I've found a workaround:

```yaml
builds:
  - id: your-id
    # Your other build options 
    tool: tinygo
    hooks:
      pre:
        # This dir needs to be made manually because TinyGo doesn't create output directories
        # like Go does. https://github.com/tinygo-org/tinygo/issues/4687
        - mkdir -p {{ dir .Path }}
```

dgryski, 2025-01-13:
I don't have time at the moment to fix this, but around here: https://github.com/tinygo-org/tinygo/blob/release/builder/build.go#L616 we'll need a call to `filepath.Dir()` to extract the directory from `outpath` and then one to `os.MkdirAll()` to ensure it exists.

jphastings, 2025-01-18:
I had a crack at fixing this with #4707 @dgryski; do let me know there if you have any advice!

deadprogram, 2025-03-04:
Closed as part of v0.36.0

PR #4707 (2025-02-19): Ensure build output directory is created
`tinygo build -o /path/to/out .` expected `/path/to/out` to already exist, which is different behaviour to `go build`. This change ensures tinygo creates any directories needed to be able to build to the specified output dir.

I wasn't certain how to effectively test this change within this repo (this is my first TinyGo contribution), any advice would be welcome!

fixes #4687