tinygo-org/tinygo #4688: Ticker.Reset is broken on RP2350

tdunning, 2025-01-05:
I wrote a small program that was using a Ticker as a timeout and found that resetting the Ticker exposed a missing bit in the runtime.

Here is a tiny program that illustrates this:

```go
package main

import (
	"time"
)

func main() {
	t := time.NewTicker(10 * time.Millisecond)
	t.Reset(time.Duration(100) * time.Millisecond)
}
```

Attempting to compile and flash this results in this output:
```bash
$  tinygo flash -target=pico2 -scheduler=tasks -gc=conservative -size=full -stack-size=20kb -monitor TickerBug/main.go
/usr/local/go/src/time/tick.go:58: linker could not find symbol time.modTimer
$
```

tdunning, 2025-01-06:
Ref to @soypat @deadprogram @aykevl @cibomahto

tdunning, 2025-01-08:
I think that this is more widespread than just the RP2350. I just tried it with -target=pico and got the same results. Tinygo never even got to flashing the board.

tdunning, 2025-01-08:
Changing to -target=nodemcu causes a very similar error message. I think that this issue is much more widespread than just the RP boards.

```
% **tinygo flash -target=nodemcu -scheduler=tasks -gc=conservative -size=full -stack-size=20kb -monitor** TickerBug/main.go
ld.lld: error: undefined symbol: time.modTimer
>>> referenced by main
>>>               /var/folders/9_/dgh8c17j1ln5q9x8jqb5x3yh0000gp/T/tinygo1875079224/main.lto.main.o:(.literal.runtime.run$1$gowrapper+0x20)
```

deadprogram, 2025-02-04:
Is this a duplicate of https://github.com/tinygo-org/tinygo/issues/3456 ?

tdunning, 2025-02-04:
Actually, I think it is. I will port my comment to that issue.

Thanks for noticing that!!

davecheney, 2026-10-24:
Duplicate of #3456

closed 2026-10-24 by davecheney