tinygo-org/tinygo #4689: GPIO interrupts not recognized on RP2350 (Pico 2)

tdunning, 2025-01-05:
I have written a simple program that should recognize edges on an input and trigger an interrupt. The interrupt doesn't happen.

Here is the simple program:
```go
package main

import (
	"fmt"
	"machine"
	"time"
)

func bcast(x bool) int {
	if x {
		return 1
	} else {
		return 0
	}
}

func main() {
	// square wave is fed to pin 10
	p := machine.Pin(10)
	p.Configure(machine.PinConfig{Mode: machine.PinInputPullup})
	// and should be echoed to pin 11
	q := machine.Pin(11)
	q.Configure(machine.PinConfig{Mode: machine.PinOutput})
	err := p.SetInterrupt(machine.PinRising, func(pin machine.Pin) {
		fmt.Printf("verify pin: %d\n", pin)
		fmt.Printf("interrupt: %d, %d\n", bcast(p.Get()), bcast(pin.Get()))
		q.Set(p.Get())
	})
	if err != nil {
		fmt.Printf("err set interrupt: %s\n", err.Error())
		panic(err.Error())
	} else {
		fmt.Printf("set interrupt ok\n")
	}
	for i := 0; i < 300; i++ {
		fmt.Printf("tick %d, pin=%d\n", i, bcast(p.Get()))
		time.Sleep(1 * time.Second)
	}
}
```
And here is the output when run on a Pico2. The main loop runs fine and the pin is clearly changing state but the interrupt is not recognized.

I suspect that there is something about the RP2350 that requires a slight additional bit of work in order to enable the control of interrupts. It may have to do with the newly introduced security model. 

I don't have a 2040 based Pico to test on right now so I can't confirm this works on the RP2040
```
 tinygo flash -target=pico2 -scheduler=tasks -gc=conservative -size=full -stack-size=20kb -monitor IrqBug/main.go
   code  rodata    data     bss |   flash     ram | package
------------------------------- | --------------- | -------
      0      45       3       5 |      48       8 | (padding)
   1560       0      10      45 |    1570      55 | (unknown)
   1220       0       0       0 |    1220       0 | C compiler-rt-builtins
      0     248       0       0 |     248       0 | C interrupt vector
     76       0       0       0 |      76       0 | C picolibc
      0       0       0    2048 |       0    2048 | C stack
    160       0       0       0 |     160       0 | Go interface assert
    292       0       0       0 |     292       0 | Go interface method
      0    1148       0       0 |    1148       0 | Go types
     96       0       0       0 |      96       0 | device/arm
     68       0       0       0 |      68       0 | device/rp
      6       0       0       0 |       6       0 | errors
   8286     216       0       0 |    8502       0 | fmt
     14       0       0       0 |      14       0 | internal/binary
   1136      21       0       0 |    1157       0 | internal/fmtsort
    132       2       0       0 |     134       0 | internal/itoa
      0      19       8       0 |      27       8 | internal/oserror
    338      48       4       4 |     390       8 | internal/task
     46       2       0       0 |      48       0 | io/fs
   1662     159       8    1082 |    1829    1090 | machine
    400       0       0       7 |     400       7 | machine/usb/cdc
      0       0      93       0 |      93      93 | machine/usb/descriptor
    440      88       0       0 |     528       0 | main
    134       0       0       0 |     134       0 | math/bits
    160      39      32       0 |     231      32 | os
   4826     773      48       0 |    5647      48 | reflect
   4350     722       4     753 |    5076     757 | runtime
    792       0       0       0 |     792       0 | runtime/volatile
    510       0       0       0 |     510       0 | sort
   7632   16012    1338       0 |   24982    1338 | strconv
     76       0       0       0 |      76       0 | sync
     32       6       0       0 |      38       0 | syscall
   1036     288       0       0 |    1324       0 | unicode/utf8
------------------------------- | --------------- | -------
  35480   19836    1548    3944 |   56864    5492 | total
Connected to /dev/cu.usbmodem14101. Press Ctrl-C to exit.
tick 1, pin=0
tick 2, pin=0
tick 3, pin=0
tick 4, pin=0
tick 5, pin=0
tick 6, pin=0
tick 7, pin=0
tick 8, pin=0
tick 9, pin=0
tick 10, pin=1
tick 11, pin=1
tick 12, pin=1
tick 13, pin=1
tick 14, pin=0
tick 15, pin=0
tick 16, pin=0
tick 17, pin=1
tick 18, pin=0
tick 19, pin=0
```

tdunning, 2025-01-06:
Ref to @soypat @deadprogram @aykevl @cibomahto

tdunning, 2025-01-08:
I have now confirmed that interrupts worked as expected on the RP2040.

That means that this is, indeed, an issue introduced by the 2350. My guess is the security stuff isn't being disabled during initial boot.

soypat, 2025-02-22:
I do not have time to look into this right now. Do consider most of existing RP2350 MCU logic was just straight up used from the existing RP2040 logic and is straight up shared, not different functions. It surely needs looking at in those nooks and crannies the new additions to the RP2350 mismatch with the RP2040

tdunning, 2025-02-22:
I might be able to find some time to look. I just don't know where to look.

Where is the startup code for the RP2040 runtime? That's probably where the 2350 specific code needs to be.

eliasnaur, 2025-03-09:
@tdunning does #4782 work for you?

tdunning, 2025-03-10:
I will give it a try when I can.

tdunning, 2025-03-11:
Hmm... I am trying to test this and the results are surprising/tricky. Part of the change in behavior may be due to the fact that I am not testing with the same exact part as before. I reviewed the changes in the reference PR and they look very plausible.

I am, however, blocked right now.  The issue I currently have is that the RPI-RP2 mount for flashing is now named RP2350. Symlinking this to a local directory with either name still results in the same issue:
```
% tinygo flash -target=pico -stack-size=30kb -monitor main.go          
failed to flash /var/folders/9_/dgh8c17j1ln5q9x8jqb5x3yh0000gp/T/tinygo649536166/main.uf2: unable to locate any volume: [RPI-RP2]
% ls -l
total 8
lrwxr-xr-x  1 tdunning  staff    15 Mar 11 09:28 RP2350 -> /Volumes/RP2350
lrwxr-xr-x  1 tdunning  staff    15 Mar 11 09:18 RPI-RP2 -> /Volumes/RP2350
-rw-r--r--  1 tdunning  staff  1410 Jan 15 09:35 main.go
% ls -l /Volumes/
total 96
drwxrwxrwx@ 1 tdunning  staff  32768 Mar  6 12:33 EOS_DIGITAL
lrwxr-xr-x  1 root      wheel      1 Mar  6 13:32 Macintosh HD -> /
drwxrwxrwx@ 1 tdunning  staff  16384 Mar 11 09:08 RP2350
%
```

eliasnaur, 2025-03-11:
FWIW, I believe you can use `tinygo build -o blah.uf2` to build an image you can copy to the rp2350 USB drive.

tdunning, 2025-03-11:
Actually, it just turns out that I am denser than a box of lead fishing weights. Five minutes after posting my tale of woe, as I was verifying that my test works on a Pico, I realized the problem.

The key was that I needed to use `-target pico2` instead of `-target pico`

PIBKAC

I have now run my test and the interrupts on the Pico2 work as expected. The world is a wonderful place.

(and I think you are correct that I could build a `uf2` binary and copy it, but that is a workaround. The real fix is more tea in the morning)

soypat, 2025-03-11:
Woohoo! Marking this issue as closed. Thank you @tdunning for the issue and @eliasnaur for the fix!

PR #4782 (2025-03-09): machine: correct register address for Pin.SetInterrupt for rp2350
Fixes #4689
commit 226744a5 (2025-03-09): machine: correct register address for Pin.SetInterrupt for rp2350 (#4782)

Fixes #4689
closed 2025-03-11 by soypat