tinygo-org/tinygo #4691: ADC in not working on bluepill

grudzinski, 2025-01-08:
Hi!

I have a simple program that reads a value from ADC:

```go
package main

import (
	"machine"
)

func main() {
	machine.InitADC()
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	sensor := machine.ADC{machine.ADC2}
	sensor.Configure(machine.ADCConfig{})
	led.High()
	val := sensor.Get()
	led.Low()
	_ = val
	for ;; {
	}
}
```

But it gets stuck on "wait for conversion to complete":

```gdb
(gdb) backtrace 
#0  0x080014ca in runtime/volatile.LoadUint32 (addr=<optimized out>) at /usr/local/lib/tinygo/src/runtime/volatile/volatile.go:25
#1  0x080014ec in (*runtime/volatile.Register32).Get (r=0x0) at /usr/local/lib/tinygo/src/runtime/volatile/register.go:142
#2  0x080014fc in (*runtime/volatile.Register32).HasBits (r=0x0, value=2) at /usr/local/lib/tinygo/src/runtime/volatile/register.go:181
#3  0x08004146 in (machine.ADC).Get (a=...) at /usr/local/lib/tinygo/src/machine/machine_stm32_adc_f1.go:67
#4  0x0800c8aa in main.main () at /home/roman/Source/github.com/grudzinski/learn-tinygo-stm32/main.go:14
```

My env:

```
roman@alienware:~$ tinygo version
tinygo version 0.35.0 linux/amd64 (using go version go1.23.4 and LLVM version 18.1.2)
```

grudzinski, 2025-01-14:
Trying to fix this myself, seems to have found some issues but it still doesn't fix it:

- [InitADC](https://github.com/tinygo-org/tinygo/blob/release/src/machine/machine_stm32_adc_f1.go#L24) is calling to [enableAltFuncClock](https://github.com/tinygo-org/tinygo/blob/release/src/machine/machine_stm32f103.go#L223) but there is no case for `stm32.RCC.SetAPB2ENR_ADC1EN(stm32.RCC_APB2ENR_ADC1EN_Enabled)` in the func. 
- Looks like ADC prescaler is not set, APB2 prescaler is set to 2, it means ADC works on 18Mhz, but max allowed is 14Mhz. `stm32.RCC.SetCFGR_ADCPRE(stm32.RCC_CFGR_ADCPRE_Div4)`

deadprogram, 2025-03-04:
Closed as part of v0.36.0

closed 2025-03-04 by deadprogram