tinygo-org/tinygo #4704: wasm: memory leak

falconandy, 2025-01-17:
Originally I met the issue in browser environment. Below is a simplified example with wazero.

tinygo version 0.35.0 linux/amd64 (using go version go1.23.5 and LLVM version 18.1.2)

Test repo: https://github.com/falconandy/tinygo-wasm-leak

Wasm lib - it just decodes png image (decoded raw image data is about 8 megabytes):
```go
package main

import (
	"bytes"
	_ "embed"
	"fmt"
	"image/png"
	"runtime"
)

//go:embed test-image.png
var testImageData []byte

//go:wasmexport processImage
func processImage() {
	img, err := png.Decode(bytes.NewReader(testImageData))
	if err != nil {
		fmt.Println(err.Error())
		return
	}

	fmt.Printf("Decoded image: width=%d height=%d\n", img.Bounds().Dx(), img.Bounds().Dy())
}

//go:wasmexport printMemUsage
func printMemUsage() {
	runtime.GC()

	var m runtime.MemStats
	runtime.ReadMemStats(&m)
	fmt.Printf("runtime.MemStats: Alloc = %v MiB, TotalAlloc = %v MiB, Sys = %v MiB\n", bToMb(m.Alloc), bToMb(m.TotalAlloc), bToMb(m.Sys))
}

func bToMb(b uint64) uint64 {
	return b / 1024 / 1024
}
```
Command to build: `tinygo build -target=wasip1 -o ./leak/leak.wasm --no-debug -scheduler=none -buildmode=c-shared ./leak`

Test code:
```go
package main

import (
	"context"
	_ "embed"
	"fmt"
	"log"
	"os"

	"github.com/tetratelabs/wazero"
	"github.com/tetratelabs/wazero/api"
	"github.com/tetratelabs/wazero/imports/wasi_snapshot_preview1"
)

//go:embed leak/leak.wasm
var leakWasm []byte

func main() {
	ctx := context.Background()

	r := wazero.NewRuntime(ctx)
	defer r.Close(ctx)

	wasi_snapshot_preview1.MustInstantiate(ctx, r)

	mod, err := r.InstantiateWithConfig(ctx, leakWasm, wazero.NewModuleConfig().WithStartFunctions("_initialize").WithStdout(os.Stdout))
	if err != nil {
		log.Fatal(err)
	}

	processImage := mod.ExportedFunction("processImage")
	printMemUsage := mod.ExportedFunction("printMemUsage")

	_, err = printMemUsage.Call(ctx)
	if err != nil {
		log.Fatal(err)
	}

	fmt.Println("mod.Memory().Size():", bToMb(mod.Memory().Size()), "MiB")

	for i := range 10 {
		fmt.Println("STEP:", i+1)
		step(ctx, mod, processImage, printMemUsage)
	}
}

func step(ctx context.Context, mod api.Module, processImage, printMemUsage api.Function) {
	_, err := processImage.Call(ctx)
	if err != nil {
		log.Fatal(err)
	}

	_, err = printMemUsage.Call(ctx)
	if err != nil {
		log.Fatal(err)
	}

	fmt.Println("mod.Memory().Size():", bToMb(mod.Memory().Size()), "MiB")
}

func bToMb(b uint32) uint32 {
	return b / 1024 / 1024
}
```

Output - memory usage only grows. A leak (10 images * 8 megabytes)?
```
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 0 MiB, Sys = 0 MiB
mod.Memory().Size(): 0 MiB
STEP: 1
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 8 MiB, TotalAlloc = 8 MiB, Sys = 15 MiB
mod.Memory().Size(): 16 MiB
STEP: 2
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 16 MiB, TotalAlloc = 16 MiB, Sys = 31 MiB
mod.Memory().Size(): 32 MiB
STEP: 3
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 24 MiB, TotalAlloc = 25 MiB, Sys = 31 MiB
mod.Memory().Size(): 32 MiB
STEP: 4
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 33 MiB, TotalAlloc = 33 MiB, Sys = 63 MiB
mod.Memory().Size(): 64 MiB
STEP: 5
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 41 MiB, TotalAlloc = 41 MiB, Sys = 63 MiB
mod.Memory().Size(): 64 MiB
STEP: 6
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 49 MiB, TotalAlloc = 50 MiB, Sys = 63 MiB
mod.Memory().Size(): 64 MiB
STEP: 7
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 58 MiB, TotalAlloc = 58 MiB, Sys = 63 MiB
mod.Memory().Size(): 64 MiB
STEP: 8
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 66 MiB, TotalAlloc = 66 MiB, Sys = 127 MiB
mod.Memory().Size(): 128 MiB
STEP: 9
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 74 MiB, TotalAlloc = 75 MiB, Sys = 127 MiB
mod.Memory().Size(): 128 MiB
STEP: 10
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 83 MiB, TotalAlloc = 83 MiB, Sys = 127 MiB
mod.Memory().Size(): 128 MiB
```

If I run the code locally (a unit test) - only 8 megabytes are "leaked" (1 image * 8 megabytes ?).
```go
func TestNativeLeak(t *testing.T) {
	printMemUsage()

	for i := range 10 {
		fmt.Println("STEP:", i+1)
		processImage()
		printMemUsage()
	}
}
```

`tinygo test -v ./leak/`

```
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 0 MiB, Sys = 0 MiB
STEP: 1
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 8 MiB, TotalAlloc = 8 MiB, Sys = 8 MiB
STEP: 2
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 8 MiB, TotalAlloc = 16 MiB, Sys = 21 MiB
STEP: 3
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 8 MiB, TotalAlloc = 25 MiB, Sys = 21 MiB
STEP: 4
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 8 MiB, TotalAlloc = 33 MiB, Sys = 21 MiB
STEP: 5
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 8 MiB, TotalAlloc = 41 MiB, Sys = 21 MiB
STEP: 6
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 8 MiB, TotalAlloc = 50 MiB, Sys = 21 MiB
STEP: 7
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 8 MiB, TotalAlloc = 58 MiB, Sys = 21 MiB
STEP: 8
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 8 MiB, TotalAlloc = 67 MiB, Sys = 21 MiB
STEP: 9
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 8 MiB, TotalAlloc = 75 MiB, Sys = 21 MiB
STEP: 10
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 8 MiB, TotalAlloc = 83 MiB, Sys = 21 MiB
--- PASS: TestNativeLeak (0.11s)
```

The unit test with Go 1.23.5 - no leaks:
```
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 0 MiB, Sys = 6 MiB
STEP: 1
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 8 MiB, Sys = 20 MiB
STEP: 2
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 17 MiB, Sys = 20 MiB
STEP: 3
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 25 MiB, Sys = 20 MiB
STEP: 4
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 33 MiB, Sys = 20 MiB
STEP: 5
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 42 MiB, Sys = 20 MiB
STEP: 6
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 50 MiB, Sys = 20 MiB
STEP: 7
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 58 MiB, Sys = 20 MiB
STEP: 8
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 67 MiB, Sys = 20 MiB
STEP: 9
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 75 MiB, Sys = 20 MiB
STEP: 10
Decoded image: width=2480 height=3508
runtime.MemStats: Alloc = 0 MiB, TotalAlloc = 84 MiB, Sys = 20 MiB
```

ydnar, 2025-03-09:
If you run it 100x or 500x, you’ll see the memory usage top out. The GC is working.

falconandy, 2025-03-10:
@ydnar 
> If you run it 100x or 500x, you’ll see the memory usage top out. The GC is working.

Probably it depends on allocated size. If I increase `N` to `10*1024*1024`, the wasm version still leaks (native 'tinygo test' - no leaks):
```
...
STEP: 201
10485760
runtime.MemStats: Alloc = 2010 MiB, TotalAlloc = 2010 MiB, Sys = 2047 MiB
mod.Memory().Size(): 2048 MiB
STEP: 202
2025/03/10 09:19:47 wasm error: out of bounds memory access
wasm stack trace:
        main.runtime.growHeap() i32
        main.runtime.alloc(i32,i32) i32
        main.main.processData#wasmexport()
exit status 1
```

`tinygo version 0.36.0 linux/amd64 (using go version go1.24.1 and LLVM version 19.1.2)`

jakebailey, 2026-09-27:
This was fixed by #5575.

closed 2026-09-27 by jakebailey