tinygo-org/tinygo #4719: Support for Go 1.24

QuLogic, 2025-01-26:
Go 1.24 is almost out; this issue records the problems building against it, mostly due to new internals.

From `go test`:
```
=== RUN   TestWasmFuncOf
    main_test.go:729: failed to build binary: wasm-ld-18: error: /tmp/TestWasmFuncOf2330703207/001/main.o: undefined symbol: internal/synctest.acquire

=== NAME  TestWasmExportJS/default
    main_test.go:770: failed to build binary: wasm-ld-18: error: /tmp/TestWasmExportJSdefault3663074247/001/main.o: undefined symbol: time.runtimeNano

=== NAME  TestBuild/EmulatedCortexM3/timers.go
    main_test.go:421: /builddir/build/BUILD/tinygo-0.35.0-build/BUILDROOT/usr/lib64/tinygo/src/runtime/scheduler_cooperative.go:251: linker could not find symbol time.runtimeNano
    main_test.go:421: /usr/lib/golang/src/time/sleep.go:56: linker could not find symbol time.runtimeNano
    main_test.go:421: /usr/lib/golang/src/time/time.go:1341: linker could not find symbol time.runtimeNow
    main_test.go:421: /usr/lib/golang/src/time/zoneinfo_read.go:323: linker could not find symbol time.runtimeNow

=== NAME  TestTest/Host/Pass
    main_test.go:988: FAIL	github.com/tinygo-org/tinygo/tests/testing/pass	0.000s
    main_test.go:897: test error: could not compile: /usr/lib/golang/src/internal/race/norace.go:37:26: undefined: abi.Type
    main_test.go:900: test failed
```
plus various variations of the above in other similar tests.

And then for `make wasmtest`:
```
+ make wasmtest
go test ./tests/wasm
--- FAIL: TestEvent (0.81s)
    setup_test.go:124: Started server at "http://127.0.0.1:44513" for dir: /tmp/TestEvent2846266819/001
    setup_test.go:28: Command: tinygo build -o /tmp/TestEvent2846266819/001/event.wasm -target wasm testdata/event.go; err=exit status 1; full output:
        wasm-ld-18: error: /tmp/tinygo2067221720/main.o: undefined symbol: internal/synctest.acquire
        wasm-ld-18: error: /tmp/tinygo2067221720/main.o: undefined symbol: internal/synctest.inBubble
    event_test.go:15: exit status 1
--- FAIL: TestFmt (0.20s)
    setup_test.go:124: Started server at "http://127.0.0.1:34557" for dir: /tmp/TestFmt858045291/001
    setup_test.go:28: Command: tinygo build -o /tmp/TestFmt858045291/001/fmt.wasm -target wasm testdata/fmt.go; err=exit status 1; full output:
        # internal/race
        /usr/lib/golang/src/internal/race/norace.go:37:26: undefined: abi.Type
        /usr/lib/golang/src/internal/race/norace.go:46:27: undefined: abi.Type
    fmt_test.go:15: exit status 1
--- FAIL: TestFmtprint (0.20s)
    setup_test.go:124: Started server at "http://127.0.0.1:43779" for dir: /tmp/TestFmtprint1890558147/001
    setup_test.go:28: Command: tinygo build -o /tmp/TestFmtprint1890558147/001/fmtprint.wasm -target wasm testdata/fmtprint.go; err=exit status 1; full output:
        # internal/race
        /usr/lib/golang/src/internal/race/norace.go:37:26: undefined: abi.Type
        /usr/lib/golang/src/internal/race/norace.go:46:27: undefined: abi.Type
    fmtprint_test.go:15: exit status 1
--- FAIL: TestLog (0.22s)
    setup_test.go:124: Started server at "http://127.0.0.1:42337" for dir: /tmp/TestLog903893515/001
    setup_test.go:28: Command: tinygo build -o /tmp/TestLog903893515/001/log.wasm -target wasm testdata/log.go; err=exit status 1; full output:
        # internal/race
        /usr/lib/golang/src/internal/race/norace.go:37:26: undefined: abi.Type
        /usr/lib/golang/src/internal/race/norace.go:46:27: undefined: abi.Type
    log_test.go:15: exit status 1
FAIL
FAIL	_/builddir/build/BUILD/tinygo-0.35.0-build/tinygo-0.35.0/tests/wasm	4.841s
FAIL
```
and with `make tinygo-test`:
```
+ make tinygo-test
/builddir/build/BUILD/tinygo-0.35.0-build/BUILDROOT/usr/bin/tinygo test cmp compress/lzw compress/zlib container/heap container/list container/ring crypto/des crypto/ecdsa crypto/elliptic crypto/md5 crypto/rc4 crypto/sha1 crypto/sha256 crypto/sha512 database/sql/driver debug/macho embed/internal/embedtest encoding encoding/ascii85 encoding/asn1 encoding/base32 encoding/base64 encoding/csv encoding/hex go/ast go/format go/scanner go/version hash hash/adler32 hash/crc64 hash/fnv html internal/itoa internal/profile math math/cmplx net/http/internal/ascii net/mail os path reflect sync testing testing/iotest text/scanner unicode unicode/utf16 unicode/utf8 unique  crypto/internal/nistec/fiat archive/zip compress/flate crypto/aes crypto/hmac debug/dwarf debug/plan9obj image io/ioutil mime mime/multipart mime/quotedprintable net net/mail net/textproto os/user regexp/syntax strconv text/tabwriter text/template/parse compress/bzip2 crypto/dsa index/suffixarray 
cannot find package "crypto/internal/nistec/fiat" in any of:
	/builddir/build/BUILD/tinygo-0.35.0-build/tinygo-0.35.0/tinygo.6O269u/tinygo/goroot-2ab90bc8762a9d8b2e4dd255397fac292d0872fd62a780073d614b72ec55541a/src/crypto/internal/nistec/fiat (from $GOROOT)
	/builddir/build/BUILD/tinygo-0.35.0-build/BUILDROOT/usr/lib64/tinygo/src/crypto/internal/nistec/fiat (from $GOPATH)
	/usr/share/gocode/src/crypto/internal/nistec/fiat
```

deadprogram, 2025-02-04:
Thanks for starting up this testing @QuLogic

gandarez, 2025-02-27:
Any update on this since go 1.24 is released?

deadprogram, 2025-03-04:
Closed as part of v0.36.0

closed 2025-03-04 by deadprogram