tinygo-org/tinygo #4721: Error upon calling _start after upgrading to tinygo version 0.35.0

GLMONTER, 2025-01-29:
I am using wasmtime and wasmtime-go version 29. I have a wasm module compiled with tinygo. Whenever I get the exported `_start` function and call it, I am getting an error. This was not an issue with tinygo version 0.34.0. The error should most likely be nil because the exit status is 0?

Code calling `_start`
```
mainFunc := module.Instance.GetExport(module.Store, "_start").Func()
_, err = mainFunc.Call(module.Store)
//error is non nil
```

Error received that should probably be nil
```
error while executing at wasm backtrace:\n    0: 0x7a4e - <unknown>!<wasm function 116>\n\nCaused by:\n    Exited with i32 exit status 0
```

mjmar01, 2025-02-08:
I'm experiencing a similar issue. When building a program for the WASI target, the `_start` function immediately exits with code 0.

This causes wazero to close the module, making it impossible to call exported functions after that. Version 0.34.0 doesn't behave like this.

Is there a compilation option to avoid this? Or can I do something in the `main` function to make it not exit without blocking?

deadprogram, 2025-02-12:
Please see PR #4734 for one possible solution to this.

deadprogram, 2025-03-04:
Closed as part of v0.36.0

closed 2025-03-04 by deadprogram