tinygo-org/tinygo #4725: ESP32 - ld.lld: error: undefined symbol: sync/atomic.runtime_procPin

jmhobbs, 2025-02-05:
I'm seeing a linker error when targeting ESP32 boards and using `sync/atomic.Value`.

```
$ tinygo build -target=esp32-mini32 main.go
ld.lld: error: undefined symbol: sync/atomic.runtime_procPin
>>> referenced by main
>>>               /var/folders/07/_t75gj1513lgy6yc3s6zdnfr0000gn/T/tinygo514386660/main.lto.main.o:(.literal.runtime.run$1$gowrapper+0x20)
ld.lld: error: undefined symbol: sync/atomic.runtime_procUnpin
>>> referenced by main
>>>               /var/folders/07/_t75gj1513lgy6yc3s6zdnfr0000gn/T/tinygo514386660/main.lto.main.o:(.literal.runtime.run$1$gowrapper+0x24)
```

```go
package main

import (
        "fmt"
        "sync/atomic"
)

func main() {
        v := &atomic.Value{}
        v.Store("test")
        fmt.Println(v.Load())
}
```

```
$ tinygo version
tinygo version 0.35.0 darwin/arm64 (using go version go1.23.4 and LLVM version 18.1.2)
```

Building to `wasm` or `feather-nrf52840` works fine.

aykevl, 2025-10-01:
We have implemented this for most architectures, but not for Xtensa (which the esp32 uses).
If you'd like to make a PR, you can add code to src/runtime/arch_xtensa.go that looks like this:

https://github.com/tinygo-org/tinygo/blob/3869f76887feef6c444308e7e1531b7cac1bbd10/src/runtime/arch_tinygoriscv.go#L21-L35

davecheney, 2026-10-24:
The missing Xtensa `sync/atomic.runtime_procPin` and `runtime_procUnpin` symbols were added in https://github.com/tinygo-org/tinygo/commit/a24dc8c543228afc8cb0fcc8e352df9e4c81433d. The confirmation above reports that the original reproducer works on ESP32 with TinyGo 0.41.1.

```go
package main

import (
    "fmt"
    "sync/atomic"
)

func main() {
    v := &atomic.Value{}
    v.Store("test")
    fmt.Println(v.Load())
}
```

```sh
tinygo build -target=esp32-mini32 main.go
tinygo build -target=esp32-coreboard-v2 main.go
```

Please open a new issue if you encounter further problems.

closed 2026-10-24 by davecheney