tinygo-org/tinygo #4727: Pico 2w support

mansguiche, 2025-02-09:
Hello, I recently started trying to use tiny go with the (relatively new) pico 2w board.

I was wondering if tiny go currently supports it and if not, if there is a plan for support in the future.

I can see build targets in the repo for pico, pico-w, and pico2 but not pico2-w
https://github.com/tinygo-org/tinygo/blob/release/targets/pico.json
https://github.com/tinygo-org/tinygo/blob/release/targets/pico-w.json
https://github.com/tinygo-org/tinygo/blob/release/targets/pico2.json

Is there a known way to build for pico2w at the moment or are there any docs someone can point me to for making a custom build target?
For context, I am new to tinygo and low level coding but understand web and golang fairly well.

deadprogram, 2025-02-23:
See https://github.com/tinygo-org/tinygo/pull/4741

deadprogram, 2025-03-04:
Closed since this was part of 0.36.0

Thanks!

closed 2025-03-04 by deadprogram

mansguiche, 2025-03-04:
Fantastic, many thanks