tinygo-org/tinygo #4740: Support built in golang version manager.

BrownNPC, 2025-02-16:
Tinygo is very picky about the name of the go compiler executable.
It requires the command to run the go compiler to be `go` but sometimes that can cause issues if the version is not supported by tinygo.

Go already has a built-in way to manage different versions of the compiler. 
https://go.dev/doc/manage-install#installing-multiple

But the compilers installed using this method have a different binary name. Example: `go1.21.3`

I propose that tiny go should be able to detect Golang versions installed using this method automatically.

inliquid, 2025-02-24:
@mattjohnsonpint if I get you right you may try to set `GOTOOLCHAIN` env variable:
```
$ GOTOOLCHAIN=go1.23.6 tinygo version
tinygo version 0.34.0 linux/amd64 (using go version go1.23.6 and LLVM version 18.1.2)
$ GOTOOLCHAIN=go1.23.5 tinygo version
tinygo version 0.34.0 linux/amd64 (using go version go1.23.5 and LLVM version 18.1.2)

```

mattjohnsonpint, 2025-02-24:
@inliquid - that works!  Thanks!

BrownNPC, 2025-04-27:
@deadprogram could this be documented?

0pcom, 2026-10-04:
The version error could point people at this. https://github.com/0magnet/tinygo/commit/4e676cc58d3a9d2564e247272d8b2beef76f7b26 adds a hint to the too-new error to set `GOTOOLCHAIN=go1.N.0`, or to use `-go-compatibility=false`. The value has to be an exact release, because the go command rejects `go1.N` and `go1.N.x`. I'll open the PR once I'm under the repository's open PR limit.