tinygo-org/tinygo #4752: compile for wasm32-unknown-unknown without wasi

berdav, 2025-02-23:
Hello, I need to compile for wasm32-unknown-unknown architecture. I'm trying to use wasm target with a WIT file like this:

```
package tinygo:test;

interface test {
  record log {
    message: string,
  }
}

world tinygo-test {
  export test;
}
```

And an empty main file
```
package main

func main() {
}
```

But it leads to an error referring to wasi preview 1.

```
$ tinygo build -target=wasm -o tgotest.wasm --wit-packag
e ./world.wit --wit-world tinygo-test main.go
error: failed to encode a component from module

Caused by:
    0: failed to decode world from module
    1: module was not valid
    2: failed to resolve import `wasi_snapshot_preview1::proc_exit`
    3: module requires an import interface named `wasi_snapshot_preview1`
`wasm-tools component new` failed: exit status 1
```

Currently I workaround this, providing a special json file like this as the target:
```
{
        "llvm-target":   "wasm32-unknown-unknown",
        "cpu":           "generic",
        "features":      "",
        "build-tags":    [
                "tinygo.wasm",
                "wasm_unknown"
        ],
        "buildmode":     "c-shared",
        "goos":          "linux",
        "goarch":        "arm",
        "linker":        "wasm-ld",
        "libc":          "wasmbuiltins",
        "rtlib":         "none",
        "gc":            "none",
        "scheduler":     "none",
        "default-stack-size": 8192,
        "cflags": [ ],
        "ldflags": [
                "--no-entry"
        ],
        "extra-files": [
                "/tmp/c-glue.c"
        ]
}
```

But it requires C files to implement functions that are still searched by tinygo (namely: `abort`, `free`, and `malloc`).

How can I avoid the use of this workaround and compile wasm without any reference to wasi (p1 or p2) or javascript requirements?

Thanks.

deadprogram, 2025-02-23:
See https://github.com/wasmvision/wasmcv for example of how to do this.

deadprogram, 2025-03-12:
Closing since this question appears to have been answered. Please reopen if needed. Thanks!

closed 2025-03-12 by deadprogram

aykevl, 2025-03-13:
To do this, use `-target=wasm-unknown` instead of `-target=wasm`.