tinygo-org/tinygo #4759: bug: cannot compile crypto code for bare metal using Go 1.24

deadprogram, 2025-02-25:
When using Go 1.24, some code with bare metal targets cannot build. The same code appears to build fine in Go 1.23.x

For example, the drivers `lorawan/atcmd` example. See https://github.com/tinygo-org/drivers/issues/744

I confirmed that this issue also applies to other code using crypto packages running on bare metal targets. For example https://github.com/hybridgroup/tinygo-tkey/pull/2

deadprogram, 2025-02-25:
```shell
$ tinygo flash -size full -target=tkey ./examples/signer/app/
go: downloading filippo.io/edwards25519 v1.1.1-0.20250211130249-04b037b40df0
# internal/syscall/unix
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at.go:15:20: undefined: syscall.BytePtrFromString
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at.go:20:25: undefined: syscall.Syscall
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at.go:29:20: undefined: syscall.BytePtrFromString
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at.go:34:26: undefined: syscall.Syscall6
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at.go:43:21: undefined: syscall.BytePtrFromString
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at.go:53:25: undefined: syscall.Syscall6
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at.go:67:20: undefined: syscall.BytePtrFromString
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at.go:72:25: undefined: syscall.Syscall6
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at_fstatat.go:14:52: undefined: syscall.Stat_t
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at_fstatat.go:16:20: undefined: syscall.BytePtrFromString
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at_fstatat.go:21:25: undefined: syscall.Syscall6
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at_sysnum_fstatat64_linux.go:11:37: undefined: syscall.SYS_FSTATAT64
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at_sysnum_linux.go:10:35: undefined: syscall.SYS_UNLINKAT
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at_sysnum_linux.go:11:35: undefined: syscall.SYS_OPENAT
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at_sysnum_linux.go:12:35: undefined: syscall.SYS_READLINKAT
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/at_sysnum_linux.go:13:35: undefined: syscall.SYS_MKDIRAT
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/copy_file_range_unix.go:15:26: undefined: syscall.Syscall6
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/faccessat_syscall.go:11:25: undefined: syscall.Faccessat
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/getrandom.go:36:26: undefined: syscall.Syscall
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/kernel_version_linux.go:15:20: undefined: syscall.Utsname
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/kernel_version_linux.go:16:20: undefined: syscall.Uname
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/nonblocking_unix.go:12:32: undefined: syscall.F_GETFL
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/nonblocking_unix.go:16:22: undefined: syscall.O_NONBLOCK
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/nonblocking_unix.go:20:22: undefined: syscall.O_NONBLOCK
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/pidfd_linux.go:10:25: undefined: syscall.Syscall
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/pidfd_linux.go:18:29: undefined: syscall.Syscall
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/tcsetpgrp_linux.go:16:25: undefined: syscall.Syscall6
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/tcsetpgrp_linux.go:16:42: undefined: syscall.SYS_IOCTL
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/tcsetpgrp_linux.go:16:82: undefined: syscall.TIOCSPGRP
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/waitid_linux.go:18:25: undefined: syscall.Syscall6
/home/ron/.gvm/gos/go1.24/src/internal/syscall/unix/waitid_linux.go:18:42: undefined: syscall.SYS_WAITID
```

jashok5, 2025-02-28:
I think I encountered the same problem. When I used tinygo to compile wasm, I got an error: go/1.24.0/libexec/src/crypto/internal/sysrand/rand_js.go:13:22: //go:wasmimport gojs runtime.getRandomData: unsupported parameter type []byte. My go version is 1.24