tinygo-org/tinygo #4765: cgo: invalid #cgo line: CXXFLAGS, pkg-config

mappu, 2025-03-01:
Hi! TinyGo is super interesting, thank you for all your work on this project.

I have [a large CGO package](https://github.com/mappu/miqt) that i'm interested to try with TinyGo, to reduce binary size and perform dead code elimination across Go/C/C++ boundaries. But, I hit some missing features so far.

```bash
$ cd miqt/examples/helloworld

$ tinygo build
# github.com/mappu/miqt/qt
../../qt/cflags.go:4:6: invalid #cgo line: CXXFLAGS
../../qt/cflags.go:6:6: invalid #cgo line: pkg-config
```

Go with CGO uses these directives:
- Any .cpp files in the same package directory are compiled using the `$(go env CXX)` compiler, using these #cgo CXXFLAGS
- #cgo pkg-config declares the .pc file to use with `$(go env PKG_CONFIG) --cflags/--libs`.

aykevl, 2025-03-01:
This is not something we currently support, since TinyGo statically links against its own libc (musl) while external libraries will most likely link against the system libc (often glibc). Linking against the wrong libc will cause all kinds of issues if it's even possible.

We would need to link to the system libc (probably) to support this. Which is something that I'd like to add support for at some point, but is a bit more work than just adding support for pkg-config.

CXXFLAGS on the other hand is something that we could definitely add support for, and wouldn't be terribly difficult.

mappu, 2025-03-01:
Thank you for the update, i understand. There's no pressure on my side.

How does GOOS=windows work? Surely that targets the system libc, not musl.

plutonium-239, 2025-07-08:
I second this feature, I'm trying to compile [`go-duckdb`](https://github.com/marcboeker/go-duckdb/).

The error:
```
# github.com/duckdb/duckdb-go-bindings/windows-amd64
...\github.com\duckdb\duckdb-go-bindings\windows-amd64@v0.1.11\cgo_static.go:4:6: invalid #cgo line: CPPFLAGS
```

env:
```env
CC: zig cc -Wno-dll-attribute-on-redeclaration -target x86_64-windows-gnu
CXX: zig c++ -target x86_64-windows-gnu
CGO_CFLAGS: -ID:\env\duckdb-libs\
CGO_LDFLAGS: -LD:\env\duckdb-libs\ -lduckdb
```

Afaik with `tinygo` the `CC`/`CXX` are ignored, they were added to show how I build this with go _'normally'_.

aykevl, 2025-09-29:
@mappu on Windows it targets UCRTBASE.DLL, which is a system library. (But Windows doesn't have an equivalent of `pkg-config`, as far as I'm aware).

@plutonium-239 we currently don't support CPPFLAGS, but it should be easy to add. As long as we only support C files, we can treat CPPFLAGS as an alias for CFLAGS. If you want, you can help make a PR by changing this code:

https://github.com/tinygo-org/tinygo/blob/3869f76887feef6c444308e7e1531b7cac1bbd10/cgo/cgo.go#L543