tinygo-org/tinygo #4773: program too large for this chip; not getting -size full dump

scottfeldman, 2025-03-04:
With v0.36.0, my program will not fit in flash, which is no problem, except I can't see what to trim because the -size full dump doesn't get printed.

```
$ tinygo build -target wioterminal -stack-size 8kb -size full runner.go
exit status 1: program too large for this chip (flash overflowed by 9052 bytes)
	optimization guide: https://tinygo.org/docs/guides/optimizing-binaries/
```

Here it would be nice to see the -size full output.

deadprogram, 2025-03-07:
Perhaps you can try to build for a target with more flash such as pico and get the output that way.

aykevl, 2025-10-01:
Yeah, this is difficult.
We need to have a linked binary to be able to print the size. But the binary is too big to be linked. So the size can't be printed.

A dirty workaround would be to modify the 0x00080000 here to be larger to produce a binary:
https://github.com/tinygo-org/tinygo/blob/3869f76887feef6c444308e7e1531b7cac1bbd10/targets/atsamd51.ld#L4
But don't forget to change it back afterwards! The linked binary won't be correct with this change.