tinygo-org/tinygo #4777: crypto/sha256 //go:wasmimport gojs runtime.getRandomData: unsupported parameter type []byte

yoursunny, 2025-03-05:
`"crypto/sha256"` is no longer importable since Go 1.24.

## go.mod

```text
module tinygo.org.localhost/demo

go 1.24.1
```

## main.go

```go
package main

import _ "crypto/sha256"

func main() {
}
```

## Go 1.24.1 + TinyGo 0.36.0: build failure

```shell
$ tinygo build -target=wasm -o main.wasm
/usr/local/go/src/crypto/internal/sysrand/rand_js.go:13:22: //go:wasmimport gojs runtime.getRandomData: unsupported parameter type []byte
```

## Go 1.23.7 + TinyGo 0.35.0 or 0.36.0: build success

1. It's necessary to modify **go.mod** with `go 1.23.7`, otherwise the compiler will automatically download Go 1.24.1.
2. **main.go** builds successfully under these combinations.

ydnar, 2025-03-06:
Does this fail in big Go as well?

ydnar, 2025-03-06:
Big Go uses a different ABI for when the module is `gojs`:

https://github.com/golang/go/blob/6a4bc8d17eb6703baf0c483fb40e0d3e1f0f6af3/src/cmd/compile/internal/ssagen/abi.go#L528-L539

deadprogram, 2025-03-06:
Any changes in the point update that impact this?

https://github.com/golang/go/issues?q=milestone%3AGo1.24.1+label%3ACherryPickApproved

aykevl, 2025-03-07:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/4784

deadprogram, 2025-03-19:
Now closing with 0.37.0 release. Thanks!

closed 2025-03-19 by deadprogram