tinygo-org/tinygo #4786: Inconsistent Behavior with `len` on Nil Array Pointer in tinygo

Yeaseen, 2025-03-07:
The `len` function behaves inconsistently when applied to a `nil` array pointer in TinyGo compared to standard Go. While Go 1.24.1 returns the array length without a panic, TinyGo produces a runtime error when dereferencing a `nil` array pointer. 

```bash
$ cat test.go
package main
func nilPtrToArray() *[2]byte {
        println("Some Activity")
        return nil
}
func main() {
        println(len(*nilPtrToArray()))

$ go version
go version go1.24.1 linux/amd64

$ go run test.go
Some Activity
2

$ tinygo run test.go
Some Activity
panic: runtime error at 0x0000000000203697: nil pointer dereference
[tinygo: panic at test.go:7:14]
failed to run compiled binary /tmp/tinygo782597653/main: signal: aborted

$ tinygo version
tinygo version 0.36.0 linux/amd64 (using go version go1.24.1 and LLVM version 19.1.2)
```

## **Expected Behavior**  
In Go, `len(*nilPtrToArray())` should not result in a panic. Instead, it should return the **length of the array type** (in this case, `2`), as the `len` function in Go does not evaluate the contents of the array but rather its **static length**. `tinygo` should replicate this behavior to maintain **compatibility**.

## **Analysis**  
The discrepancy likely stems from how `tinygo` handles **nil pointer dereferencing** in combination with `len`. In standard Go, dereferencing a `nil` pointer to an array solely for evaluating its length does not trigger a **runtime error**. `tinygo`, however, appears to **strictly enforce pointer validity**, leading to a **nil pointer dereference panic**.

Yeaseen, 2025-03-07:
I think `cap` function also shows the same discrepancy.  

```go
func main() {
	println(cap(*nilPtrToArray()))
}
```

aykevl, 2025-03-13:
I can confirm this issue.

aykevl, 2025-03-13:
Looks like this is actually a bug in the golang.org/x/tools/go/ssa package. I'll see whether I can create a reproducer.

Yeaseen, 2025-03-13:
@aykevl It looks like a Go compiler bug! Thank you for submitting the issue on upstream.

davecheney, 2026-10-24:
TinyGo was correct here. The Go compiler was incorrectly skipping the nil dereference, which was fixed in Go 1.25: https://github.com/golang/go/issues/72844. Go now agrees with TinyGo.

This program demonstrates the distinction:

```go
package main

import "fmt"

func nilArray() *[2]byte {
    fmt.Println("nilArray called")
    return nil
}

func main() {
    p := nilArray()
    fmt.Println("len(*p):", len(*p))

    defer func() {
        if r := recover(); r != nil {
            fmt.Println("len(*nilArray()) panicked:", r)
            return
        }
        panic("expected len(*nilArray()) to panic")
    }()
    fmt.Println("len(*nilArray()):", len(*nilArray()))
}
```

```sh
go run main.go
tinygo run main.go
```

Checked with Go 1.27.3 and TinyGo 0.44.0-dev-eae3f875 on darwin/arm64. Both produce:

```text
nilArray called
len(*p): 2
nilArray called
len(*nilArray()) panicked: runtime error: invalid memory address or nil pointer dereference
```

`len(*p)` is a constant expression, so `*p` is not evaluated. The function call inside `len(*nilArray())` makes that expression nonconstant, requiring evaluation and the nil dereference.

Closing because the reported TinyGo behavior is correct.

closed 2026-10-24 by davecheney