tinygo-org/tinygo #4790: https://pkg.go.dev/reflect#Value.SetIterKey missing

pjebs, 2025-03-09:
I am trying to build an app on tinygo.

It spits out this error:

```go
../../go/pkg/mod/github.com/fxamacker/cbor/v2@v2.7.0/encode_map.go:33:10: iterk.SetIterKey undefined (type *reflect.Value has no field or method SetIterKey)
../../go/pkg/mod/github.com/fxamacker/cbor/v2@v2.7.0/encode_map.go:34:10: iterv.SetIterValue undefined (type *reflect.Value has no field or method SetIterValue)
../../go/pkg/mod/github.com/fxamacker/cbor/v2@v2.7.0/encode_map.go:48:9: iterk.SetIterKey undefined (type *reflect.Value has no field or method SetIterKey)
../../go/pkg/mod/github.com/fxamacker/cbor/v2@v2.7.0/encode_map.go:49:9: iterv.SetIterValue undefined (type *reflect.Value has no field or method SetIterValue)
```

https://pkg.go.dev/reflect#Value.SetIterKey was added in Go1.18 and https://tinygo.org/docs/reference/lang-support/stdlib/ makes no mention of the function being missing.

How can I solve this?

pjebs, 2025-03-09:
Any a few more from standard library that is not mentioned in the documentation:

```go
# github.com/valyala/fasthttp
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/client.go:1919:9: c.Clone undefined (type *tls.Config has no field or method Clone)
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/http.go:2264:13: w.ReadFrom undefined (type *os.File has no field or method ReadFrom)
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/http.go:2278:13: w.ReadFrom undefined (type *net.TCPConn has no field or method ReadFrom)
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/peripconn.go:50:7: undefined: tls.Conn
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/peripconn.go:58:31: undefined: tls.Conn
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/server.go:1613:38: tlsConn.ConnectionState().NegotiatedProtocol undefined (type tls.ConnectionState has no field or method NegotiatedProtocol)
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/server.go:1646:14: undefined: os.Chmod
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/server.go:1709:14: s.TLSConfig.BuildNameToCertificate undefined (type *tls.Config has no field or method BuildNameToCertificate)
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/server.go:1714:35: s.TLSConfig.Clone undefined (type *tls.Config has no field or method Clone)
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/server.go:1737:14: s.TLSConfig.BuildNameToCertificate undefined (type *tls.Config has no field or method BuildNameToCertificate)
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/server.go:1742:35: s.TLSConfig.Clone undefined (type *tls.Config has no field or method Clone)
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/server.go:1772:19: undefined: tls.X509KeyPair
../../go/pkg/mod/github.com/valyala/fasthttp@v1.59.0/tcpdialer.go:468:18: undefined: net.DefaultResolver
```

mikesmitty, 2025-03-15:
For the cbor package, I ran into that the other day. There's a tinygo-specific branch that's compatible. The slash in the branch name causes go get to fail unless you set GOPROXY=direct though: `GOPROXY=direct go get github.com/fxamacker/cbor/v2@feature/cbor-tinygo-beta`

PR #4814 (2025-03-17): reflect, internal/reflectlite: add Value.SetIter{Key,Value} and MapIter.Reset
Fixes #4790. Depends on #4787 (merge that first).
commit 6a25fd49 (2025-03-17): reflect, internal/reflectlite: add Value.SetIter{Key,Value} and MapIter.Reset

Fixes #4790. Depends on #4787 (merge that first).
closed 2025-03-19 by deadprogram