tinygo-org/tinygo #4793: The sync.Mutex Struct Has Different Sizes

Tryanks, 2025-03-10:
Short Description: The size of the struct measured using `unsafe.SizeOf(sync.Mutex{})` differs between Go and TinyGo.  
Quick Reproduction:  
```go  
package main
import (
    "fmt"
     "sync"
     "unsafe" 
)  
func main() {
     fmt.Println(unsafe.Sizeof(sync.Mutex{}))
}
```
In Go 1.24.1, it is 8, while in TinyGo 0.36.0, it is 16.

Additional Information:
This issue prevents me from building [cznic/sqlite](https://gitlab.com/cznic/sqlite) with TinyGo, as it results in the following error during the build:
```
# modernc.org/libc
../../../../go/pkg/mod/modernc.org/libc@v1.61.13/pthread_musl.go:43:5: unsafe.Sizeof(Tpthread_mutex_t{}) - unsafe.Sizeof(pthreadMutex{}) (constant -8 of type uintptr) overflows uintptr
```

Upon closer inspection of the code, it appears that the original libc (a pure Go implementation) compares two structs to ensure there is enough memory for struct mapping:
```go
// Source Code: https://gitlab.com/cznic/libc/-/blob/master/pthread_musl.go?ref_type=heads#L43

// C version is 40 bytes (64b) and 24 bytes (32b).
type pthreadMutex struct { // 64b        32b
	sync.Mutex            //  0	8    0     8
	count      int32      //  8	4    8     4
	mType      uint32     // 12	4   12     4
	outer      sync.Mutex // 16	8   16     8
	owner      int32      // 24	4   24     4
	//			 28         28
}

type Tpthread_mutex_t = struct {
	F__u struct {
		F__vi [0][10]int32
		F__p  [0][5]uintptr
		F__i  [10]int32
	}
}

const _ [unsafe.Sizeof(Tpthread_mutex_t{}) - unsafe.Sizeof(pthreadMutex{})]byte
```

I suspect this is due to different implementations of Mutex in TinyGo and Golang. Therefore, I am not entirely sure if this is a bug.

My Operating System:
- Macbook M2 (arm64)
- MacOS 15.3.1 (24D70)
- Go 1.24.1
- TinyGo 0.36.0

aykevl, 2025-03-10:
This is not a bug in TinyGo. Instead, the cznic/sqlite package depends on implementation details of the sync.Mutex type. It could break with any Go version update, and is therefore inherently unsafe.

I recommend filing a bug with cznic/sqlite to not depend on an implementation detail of sync.Mutex.

Tryanks, 2025-03-10:
Thank you very much for your response. Since it may become invalid at any time with the upgrade of the Go version, it indeed does not belong to the issue of TinyGo.

closed 2025-03-10 by Tryanks