tinygo-org/tinygo #4800: freebsd build got error

runsys, 2025-03-12:
root@fbsdvp:/mnt/work/code/go/src/github.com/tinygo-org/tinygo # export CPPFLAGS=-I/usr/local/llvm18/include root@fbsdvp:/mnt/work/code/go/src/github.com/tinygo-org/tinygo # go build # tinygo.org/x/go-llvm /mnt/work/code/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20240518103902-697964f2a9dc/analysis.go:16:10: fatal error: 'llvm-c/Analysis.h' file not found 16 | #include "llvm-c/Analysis.h" // If you are getting an error here you need to build or install LLVM, see https://tinygo.org/docs/guides/build/ | ^~~~~~~~~~~~~~~~~~~ 1 error generated. root@fbsdvp:/mnt/work/code/go/src/github.com/tinygo-org/tinygo #

aykevl, 2025-03-13:
Please take a look at how to set CGo variables [in the docs](https://pkg.go.dev/cmd/cgo). In particular:

> When building, the CGO_CFLAGS, CGO_CPPFLAGS, CGO_CXXFLAGS, CGO_FFLAGS and CGO_LDFLAGS environment variables are added to the flags derived from these directives. Package-specific flags should be set using the directives, not the environment variables, so that builds work in unmodified environments. Flags obtained from environment variables are not subject to the security limitations described above.

I think you need to use `CGO_CPPFLAGS` instead.

Please note that FreeBSD isn't really supported at the moment. It might work, or it might not. But we don't test for it.

davecheney, 2026-10-26:
Thanks for the report. Please use `CGO_CPPFLAGS` rather than `CPPFLAGS`, as explained above. FreeBSD remains an untested platform, but additional bug reports are welcome.

closed 2026-10-26 by davecheney