tinygo-org/tinygo #4804: fail to build go-json-experiment

Gaubee, 2025-03-13:
```
# github.com/go-json-experiment/json
../../../go/pkg/mod/github.com/go-json-experiment/json@v0.0.0-20250223041408-d3c622f1b874/arshal_default.go:784:8: k.SetIterKey undefined (type addressableValue has no field or method SetIterKey)
../../../go/pkg/mod/github.com/go-json-experiment/json@v0.0.0-20250223041408-d3c622f1b874/arshal_default.go:798:8: v.SetIterValue undefined (type addressableValue has no field or method SetIterValue)
../../../go/pkg/mod/github.com/go-json-experiment/json@v0.0.0-20250223041408-d3c622f1b874/arshal_default.go:806:8: k.SetIterKey undefined (type addressableValue has no field or method SetIterKey)
../../../go/pkg/mod/github.com/go-json-experiment/json@v0.0.0-20250223041408-d3c622f1b874/arshal_default.go:834:8: k.SetIterKey undefined (type addressableValue has no field or method SetIterKey)
../../../go/pkg/mod/github.com/go-json-experiment/json@v0.0.0-20250223041408-d3c622f1b874/arshal_default.go:836:8: v.SetIterValue undefined (type addressableValue has no field or method SetIterValue)
../../../go/pkg/mod/github.com/go-json-experiment/json@v0.0.0-20250223041408-d3c622f1b874/arshal_inlined.go:133:8: mk.SetIterKey undefined (type addressableValue has no field or method SetIterKey)
../../../go/pkg/mod/github.com/go-json-experiment/json@v0.0.0-20250223041408-d3c622f1b874/arshal_inlined.go:145:8: mk.SetIterKey undefined (type addressableValue has no field or method SetIterKey)
```

Syuparn, 2025-03-17:
It seems to be related to #4790 (`reflect#Value.SetIterKey` is missing). 

method `SetIterKey` and `SetIterValue` come from `addressableValue`'s embbeded field `reflect.Value`.

https://github.com/go-json-experiment/json/blob/d3c622f1b874954c355e60c8e6b6baa5f60d2fed/arshal_default.go#L784

https://github.com/go-json-experiment/json/blob/d3c622f1b874954c355e60c8e6b6baa5f60d2fed/arshal.go#L496-L503

sxzz, 2025-03-18:
Tried [0.37.0-dev-e5a8d693](https://github.com/tinygo-org/tinygo/commit/e5a8d693), `SetIterKey` and `SetIterValue` methods are resolved, but still remains some issues
```
tinygo version 0.37.0-dev-e5a8d693 darwin/arm64 (using go version go1.24.1 and LLVM version 19.1.2)

# github.com/go-json-experiment/json
../../../go/pkg/mod/github.com/go-json-experiment/json@v0.0.0-20250223041408-d3c622f1b874/arshal_default.go:829:39: undefined: reflect.SliceOf
../../../go/pkg/mod/github.com/go-json-experiment/json@v0.0.0-20250223041408-d3c622f1b874/arshal_default.go:830:39: undefined: reflect.SliceOf
../../../go/pkg/mod/github.com/go-json-experiment/json@v0.0.0-20250223041408-d3c622f1b874/arshal_default.go:927:36: undefined: reflect.MapOf
```

PR #4836 (2025-04-04): reflect: Add SliceOf, ArrayOf, StructOf, MapOf, FuncOf
Fixes #4804. Adds helpers pointed to internal/reflectlite to allow building. Underlying implementation is still missing.
closed 2025-06-16 by deadprogram