tinygo-org/tinygo #4807: html/template: reflect Seq/CanSeq/CanSeq2 undefined with go1.24.1; works with go1.23.4

scottfeldman, 2025-03-14:
Getting reflect compile errors on code using html/template after upgrading to go1.24.1.

```
$ go version
go version go1.24.1 linux/amd64

$ tinygo version
tinygo version 0.37.0-dev-4f7c64cb linux/amd64 (using go version go1.24.1 and LLVM version 17.0.6)

$ tinygo build -target nano-rp2040 ./main.go 
# text/template
/usr/local/go/src/text/template/exec.go:405:22: val.Seq undefined (type reflect.Value has no field or method Seq)
/usr/local/go/src/text/template/exec.go:454:17: val.Type().CanSeq undefined (type reflect.Type has no field or method CanSeq)
/usr/local/go/src/text/template/exec.go:460:23: val.Seq undefined (type reflect.Value has no field or method Seq)
/usr/local/go/src/text/template/exec.go:471:17: val.Type().CanSeq2 undefined (type reflect.Type has no field or method CanSeq2)
/usr/local/go/src/text/template/exec.go:473:26: val.Seq2 undefined (type reflect.Value has no field or method Seq2)
```

If I revert back to go1.23.4, then compile works:

```
$ go version
go version go1.23.4 linux/amd64

$ tinygo build -target nano-rp2040 ./main.go 
```

Test program is:

```
package main

import (
	"log"
	"os"
	"text/template"
	"time"
)

type PageData struct {
	Title   string
	Message string
	Name    string
}

const htmlTemplate = `
<!DOCTYPE html>
<html>
<head>
    <title>{{.Title}}</title>
</head>
<body>
    <h1>{{.Title}}</h1>
    <div class="message">
        {{.Message}}
        {{if .Name}}
            <p>Welcome, {{.Name}}!</p>
        {{else}}
            <p>Please provide a name in the URL query!</p>
        {{end}}
    </div>
</body>
</html>
`

func main() {
	time.Sleep(2 * time.Second)

	// Parse the template
	tmpl, err := template.New("webpage").Parse(htmlTemplate)
	if err != nil {
		log.Fatal("Error parsing template:", err)
	}

	// Create data for the template
	data := PageData{
		Title:   "Template Demo",
		Message: "This is a demonstration of Go's html/template package",
		Name:    "Joe",
	}

	// Execute the template with the data
	err = tmpl.Execute(os.Stdout, data)
	if err != nil {
		log.Println("Error executing template:", err)
		return
	}

	for {
		time.Sleep(time.Second)
	}
}
```

ydnar, 2025-03-17:
This is fixed in #4806.

deadprogram, 2025-03-19:
Now closing with 0.37.0 release. Thanks!

closed 2025-03-19 by deadprogram