tinygo-org/tinygo #4810: builder: runtime.buildVersion not a string

ydnar, 2025-03-17:
When running tests under `-target=wasip2`, the build for #4809 was failing when testing `crypto/elliiptic`:

```console
$ tinygo test -v -target=wasip2 crypto/elliptic
FAIL	crypto/elliptic	0.000s
runtime.buildVersion: not a string
```

I traced the issue to an error type in package `internal/cm`:

```go
var (
	errEmpty          = &stringError{err: "empty text"}
	errNoMatchingCase = &stringError{err: "no matching case"}
)

type stringError struct {
	_   struct{}
	err string
}

func (err *stringError) Error() string {
	return err.err
}
```

Removing this type fixes the build-time error.

aykevl, 2025-03-17:
That's definitely not supposed to happen!
How can I reproduce this?

aykevl, 2025-03-17:
Oh I have a reproducer.
At commit afd4dc222e2eef0d751de4d08e8ef103511fa033, I got:

```
$ tinygo run -target=wasip2 ./testdata/alias.go
runtime.buildVersion: not a string
```

aykevl, 2025-03-17:
Huh. It looks like LLVM has decided that `{ ptr, i32 }` and `%runtime._string` are the same type. And they do have the same layout in memory. But this is a bit unexpected.

aykevl, 2025-03-17:
This fixes the bug: https://github.com/tinygo-org/tinygo/pull/4811

closed 2025-03-19 by deadprogram