tinygo-org/tinygo #4815: stdlib/sync: Match Mutex size with upstream Go

soypat, 2025-03-17:
People are trying to build stuff that uses [modernc.org/memory package](https://gitlab.com/cznic/memory).

The current implementation of sync.Mutex is based on the task.Mutex which is composed of a bool and a Stack (pointer). This gives the Mutex a size of 16 bytes on 64-bit aligned systems while in upstream Go it is 8 bytes.

### TinyGo data structures
- [sync.Mutex](https://github.com/tinygo-org/tinygo/blob/8c54e3dd885824638015f319907f75c01570cdad/src/sync/mutex.go#L7)
- [task.Mutex](https://github.com/tinygo-org/tinygo/blob/8c54e3dd885824638015f319907f75c01570cdad/src/internal/task/mutex-cooperative.go#L3)
- [task.Stack](https://github.com/tinygo-org/tinygo/blob/8c54e3dd885824638015f319907f75c01570cdad/src/internal/task/queue.go#L72)

### Go data structures 
- [sync.Mutex](https://github.com/golang/go/blob/c7ea87132f4e6f3c81e525c396a64471c9af0091/src/sync/mutex.go#L30)
- [isync.Mutex](https://github.com/golang/go/blob/c7ea87132f4e6f3c81e525c396a64471c9af0091/src/internal/sync/mutex.go#L20)

aykevl, 2025-03-18:
To be honest, this is a bug in [modernc.org/memory](https://gitlab.com/cznic/memory), not in TinyGo. They have written non-portable code. If they want to extend portability to TinyGo they could use a build tag. But it's also possible we're going to break this at some point - for example, `-scheduler=threads` and `-scheduler=tasks` may have a different `sync.Mutex` size.

I don't think there's anything to fix on our side.

Tryanks, 2025-03-20:
@soypat 
In the commit for [libc_issues#43](https://gitlab.com/cznic/libc/-/issues/43#note_2403463049), the allocated memory has been expanded in an attempt to make it compatible with the current tinygo's sync.Mutex. If it works, then this issue seems to have been resolved.

However, after fixing this issue, a new problem has emerged:
```
i64 add (i64 ptrtoint (ptr @"modernc.org/libc._table" to i64), i64 256)
```

aykevl, 2025-09-30:
Closing since I don't believe this is a TinyGo bug.

@Tryanks can you make a separate issue with steps to reproduce the problem you encountered?

Tryanks, 2025-11-11:
> [@Tryanks](https://github.com/Tryanks) can you make a separate issue with steps to reproduce the problem you encountered?

Sorry, I cannot reproduce this error alone because the discrepancies seem to be increasing:
```
~/GolandProjects/TinySQLiteTest via go v1.25.4 
x tinygo run .
# modernc.org/sqlite
../../go/pkg/mod/modernc.org/sqlite@v1.40.0/fcntl.go:27:16: undefined: runtime.Pinner
```