tinygo-org/tinygo #4816: TinyGo Panics on Nil Pointer Dereference in Map Lookup, Unlike Standard Go

Yeaseen, 2025-03-18:
TinyGo incorrectly **panics on a nil pointer dereference when accessing a struct field from a map lookup**, while **standard Go handles the same code without errors**. This inconsistency suggests that **TinyGo does not optimize nil pointer access in maps the same way as the Go compiler does**.

```bash
$ cat test.go
package main
import "fmt"
type thing struct {
        field int
}
func main() {
        m := map[string]*thing{}
        t, ok := m["abc"]
        valid := t.field >= 0

        if !ok || !valid {
                fmt.Printf("got: %v\n", t)
                t = &thing{field: 0}
                m["abc"] = t
        }

        t.field += 3
        fmt.Println(t.field)
}
```

```bash
$ go version
go version go1.24.1 linux/amd64

$ go run test.go
got: <nil>
3
```

```bash
$ tinygo version
tinygo version 0.36.0 linux/amd64 (using go version go1.24.1 and LLVM version 19.1.2)

$ tinygo run test.go
panic: runtime error at 0x000000000021261d: nil pointer dereference
[tinygo: panic at /usr/local/lib/tinygo/src/runtime/scheduler_cooperative.go]
failed to run compiled binary /tmp/tinygo4289749700/main: signal: aborted
```

aykevl, 2025-03-18:
Here you go: https://github.com/golang/go/issues/72918
Pretty sure this is actually a bug in the Go compiler.

...how do you find these edge cases? This is the second one in a short time.

aykevl, 2025-03-18:
Ok, this is indeed a bug in the Go compiler. Not a TinyGo bug.

The reproducer is very similar to https://github.com/golang/go/issues/72860. I guess it originated from the same source?

closed 2025-03-18 by aykevl

Yeaseen, 2025-03-19:
@aykevl Yeah. I believe so. I just found the discrepancy in the outputs between tinygo and go.