tinygo-org/tinygo #4819: TinyGo Panics on Generic Type Aliases

Yeaseen, 2025-03-19:
TinyGo crashes when compiling a **generic type alias**, even when using `GODEBUG=gotypesalias=1`. Though I am not sure whether TinyGo supports generic type aliases correctly. 

```bash
$ cat test.go
package main
import "fmt"
// Define a generic type
type Wrapper[T any] struct {
	Value T
}
// Alias declaration with type parameter
type Alias[T any] = Wrapper[T]

func main() {
	// Using the alias
	var a Alias[int] = Alias[int]{Value: 42}
	fmt.Println("a:", a.Value)
}
```

```
GODEBUG=gotypesalias=1 tinygo run test.go
panic: unknown type: main.Alias[int]

goroutine 200 [running]:
github.com/tinygo-org/tinygo/compiler.(*compilerContext).makeLLVMType(0xc003bb2000?, {0x7020428?, 0xc00287d340?})
        /__w/tinygo/tinygo/compiler/compiler.go:468 +0x539
```

gandarez, 2025-03-20:
It produces different errors when setting target.

Without target set
```
GODEBUG=gotypesalias=1 tinygo run main.go

panic: unknown type: sync.Mutex

goroutine 54 [running]:
github.com/tinygo-org/tinygo/compiler.(*compilerContext).makeLLVMType(0x14001ca8000?, {0x10aa97468?, 0x1400587a800?})
        /Users/runner/work/tinygo/tinygo/compiler/compiler.go:468 +0x578
```

With pico2-w target set
```
GODEBUG=gotypesalias=1 tinygo run -target=pico2-w main.go

panic: unknown type: any

goroutine 116 [running]:
github.com/tinygo-org/tinygo/compiler.(*compilerContext).makeLLVMType(0x1400af8a280?, {0x10b4ab468?, 0x140000221c0?})
        /Users/runner/work/tinygo/tinygo/compiler/compiler.go:468 +0x578
```

eliasnaur, 2025-03-24:
Probable dupe: https://github.com/tinygo-org/tinygo/issues/4416

aykevl, 2025-05-24:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/4911