tinygo-org/tinygo #4822: Support for net/http Packages at Compile time

davidmdm, 2025-03-20:
TinyGo currently does not support `net/http/*` packages as documented [here](https://tinygo.org/docs/reference/lang-support/stdlib/#nethttp).

This is unfortunate as use-cases exist where we would want to import packages that depend on `net/http/*` but do not need to run networking.

For example in the Kubernetes space, we would like to write programs that depend on `k8s.io/api`, `apimachinery` and third-party operator types. Unfortunately these packages depend on `net/http` packages and we cannot compile them using tinygo.

Resulting in errors such as:
```
../../../../go/pkg/mod/golang.org/x/net@v0.37.0/internal/httpcommon/request.go:11:2: package net/http/httptrace is not in std (/Users/davidmdm/Library/Caches/tinygo/goroot-844030248698ae6eab7078512b4fefd9cf5525da9617ec2f86276fb6aa7030bc/src/net/http/httptrace)
```

The ask is to implement/stub/mock enough of net/http to allow tinygo to work with the kubernetes ecosystem at least at the type level.
The ask is NOT to implement networking or http.

#### Related issues:
- https://github.com/tinygo-org/tinygo/issues/4186
- https://github.com/tinygo-org/tinygo/issues/4598

davidmdm, 2025-03-20:
This would imply that the following program would compile successfully:

```go
package main

import (
	"fmt"

	metav1 "k8s.io/apimachinery/pkg/apis/meta/v1"
)

func main() {
	fmt.Println(metav1.ObjectMeta{})
}
```

b0ch3nski, 2025-03-21:
@davidmdm You can take a look at this: https://github.com/kubewarden/k8s-objects

davidmdm, 2025-03-21:
@b0ch3nski I am aware of that repo. In some cases it is good enough, but as soon as you want to depend on types found in third party operators you are back to square one.

glycerine, 2025-05-13:
@davidmdm 
@scottfeldman 
@aykevl 

I'm interested in this. Short backstory:

I would like to create a deterministic, single-threaded, Go execution environment such as that used in FoundationDB's Flow C++ DSL, with the similar aim of reproducible testing of fault scenarios on a simulated network. For example, my tests will randomly isolate nodes, partition the network, kill nodes, restart nodes, and have isolated nodes return from isolation and possibly disrupt the existing cluster by "coming back" from the dead.

TinyGo is appealing to me in that it provides a single threaded runtime by default. This would make it a much more suitable that Big Go because in Big Go the threads inherently introduce non-determinism, making tests that depend on the exact timing of thread scheduling difficult to reproduce. Also there is no way on Big Go to easily make the runtime use a deterministic pseudo random number generator for selects or map iteration order, but these are pre-requisites for my aims.

As a first step, I have written a network simulator completely in Go.
https://github.com/glycerine/gosimnet mocks a network without talking to an operating system. It uses only goroutines and channels to implement net.Conn. It builds and runs a basic test under TinyGo and optionally with the new testing/synctest fake-time facility. 
~~~
~/go/src/github.com/glycerine/gosimnet (master) $ tinygo test -v
faketime = false
=== RUN   Test101_gosimnet_basics

 2025-05-13 05:00:50.694 +0000 UTC simnet.Start: faketime = false; s.barrier=true
--- PASS: Test101_gosimnet_basics (1.03s)
PASS
ok      github.com/glycerine/gosimnet   1.438s
~/go/src/github.com/glycerine/gosimnet (master) $ tinygo version
tinygo version 0.37.0 darwin/amd64 (using go version go1.24.3 and LLVM version 19.1.2)
~~~
So gosimnet is a network mock, so much more than was asked for above, but perhaps the effort is complimentary, and could be incorporated or leveraged, or simply helpful in testing, the efforts to finish out the net package for TinyGo.

I too have noticed the limitations of TinyGo's current net support (net.Interface.Addrs() not being implemented, for example) when I tried to build my rpc package (the next layer on top of gosimnet) with TinyGo, and I would like to help address these missing parts.

I think that lays out my goals and current progress towards them. I would like to hear if there are unforeseen impediments. Are there other sources of randomness in the TinyGo runtime or Garbage collector, for instance?

And so, I would be open to collaborating on, or simply driving, further net support in TinyGo. 

It sounds like some work has already been done. I would be happy to build on such work if the authors feel it appropriate, or I would be fine with keeping my effort under a independent with appropriate flags/switches/tags. I would need some starting guidance on what build tags/flags would be appropriate to add to TinyGo, if any, from the project maintainers. Perhaps a simple GOOS=gosimnet would suffice; really I don't have enough context to know what is appropriate here, and I'm happy to have guidance. My hardware for testing is amd64 so that target would be my principal one, but hopefully everything can stay architecture neutral.