tinygo-org/tinygo #4828: linker could not find symbol ...

jakebailey, 2025-03-24:
I'm trying to build a version of the Go port of the TypeScript compiler with tinygo. The code is here: https://github.com/microsoft/typescript-go/tree/main/cmd/tsgo

When I just do a regular build, I get:

```console
$ ~/tinygo/bin/tinygo build -o tsgo-tinygo ./cmd/tsgo
/home/jabaile/tinygo/src/internal/futex/futex_linux.c:24: linker could not find symbol syscall
../../../../usr/local/lib/tinygo/lib/bdwgc/misc.c:896: linker could not find symbol strtoul
../../../../usr/local/lib/tinygo/lib/bdwgc/misc.c:1037: linker could not find symbol open
../../../../usr/local/lib/tinygo/lib/bdwgc/misc.c:1128: linker could not find symbol atol
../../../../usr/local/lib/tinygo/lib/bdwgc/misc.c:1139: linker could not find symbol atoi
../../../../usr/local/lib/tinygo/lib/bdwgc/misc.c:1148: linker could not find symbol atol
../../../../usr/local/lib/tinygo/lib/bdwgc/misc.c:1160: linker could not find symbol atoi
../../../../usr/local/lib/tinygo/lib/bdwgc/misc.c:1173: linker could not find symbol atoi
../../../../usr/local/lib/tinygo/lib/bdwgc/os_dep.c:235: linker could not find symbol open
../../../../usr/local/lib/tinygo/lib/bdwgc/os_dep.c:1170: linker could not find symbol open
../../../../usr/local/lib/tinygo/lib/bdwgc/os_dep.c:1200: linker could not find symbol strtoul
../../../../usr/local/lib/tinygo/lib/bdwgc/os_dep.c:2916: linker could not find symbol strtoul
../../../../usr/local/lib/tinygo/lib/bdwgc/os_dep.c:3849: linker could not find symbol fcntl
../../../../usr/local/lib/tinygo/lib/bdwgc/os_dep.c:3946: linker could not find symbol uname
```

But, I definitely have glibc installed such that those should be found.

Am I just "holding it wrong", and there's a different way to build a plain binary for my system?

aykevl, 2025-03-25:
A `tinygo clean` should fix this.

(We should eventually really fix this issue, but for now `tinygo clean` is a workaround - essentially, you have a stale cache).

ben-krieger, 2025-03-26:
`tinygo clean` fixed some of my missing symbols (strtoul, atol, etc. - same as OP). However, there are two that remain and seem to be related to Go 1.24 changes:

```console
[nix...]-go-1.24.1/share/go/src/crypto/internal/sysrand/rand.go:48: linker could not find symbol crypto/internal/sysrand.fatal
[nix...]-tinygo-0.37.0/share/tinygo/src/internal/syscall/unix/getrandom.go:26: linker could not find symbol runtime.vgetrandom
```

aykevl, 2025-03-29:
Can you share the code that is causing these errors, so I can reproduce this?

jakebailey, 2025-03-29:
It was https://github.com/microsoft/typescript-go/tree/main/cmd/tsgo, but I haven't seen this since I cleaned, so I'm not entirely certain how to get into the bad state. I think it happened when I upgraded from 0.36 to 0.37?

aykevl, 2025-03-31:
Hmm, in that case I think there's nothing to fix.

closed 2025-03-31 by aykevl

jakebailey, 2025-03-31:
Er, well, as you said it should eventually be fixed 😅

Not sure if you have a tracking issue elsewhere for it, though.

soypat, 2025-04-17:
This issue popped up with tinybench and tinygo clean is not fixing it.
- Program: https://github.com/tinygo-org/tinybench/blob/main/rsa-keygen/go/main.go

```
$ tinygo version
tinygo version 0.38.0-dev-dcf609de linux/amd64 (using go version go1.24.2 and LLVM version 19.1.2)

$ go test -v -bench=BenchmarkAll/rsa-keygen:args=-s_512/go/tinygo
goos: linux
goarch: amd64
pkg: tinybench
cpu: 12th Gen Intel(R) Core(TM) i5-12400F
BenchmarkAll
    bench_test.go:76: found compiler "go"
    bench_test.go:76: found compiler "tinygo"
    bench_test.go:76: found compiler "gcc"
    bench_test.go:78: skipping all benchmarks for compiler "clang"
    bench_test.go:78: skipping all benchmarks for compiler "zig"
    bench_test.go:81: looking for benchmarks in [fannkuch-redux n-body n-body-nosqrt rsa-keygen]
BenchmarkAll/rsa-keygen:args=-s_512/go/tinygo
    bench_test.go:111: rsa-keygen: building with tinygo flags=[build -opt=2 -o=tinybin ./rsa-keygen/go/main.go]:
        /home/pato/Documents/src/tg/tinygo/src/internal/syscall/unix/getrandom.go:26: linker could not find symbol runtime.vgetrandom
        /usr/local/go/src/crypto/internal/sysrand/rand.go:48: linker could not find symbol crypto/internal/sysrand.fatal
--- FAIL: BenchmarkAll/rsa-keygen:args=-s_512/go/tinygo
--- FAIL: BenchmarkAll
FAIL
exit status 1
FAIL    tinybench       8.973s
```

aykevl, 2025-04-17:
@soypat that's a separate issue, please make a new issue for that, with steps to reproduce.