tinygo-org/tinygo #4834: "Call parameter type does not match function signature" when trying to compile.

DagAgren, 2025-03-27:
Experimenting with Go and Wasm for the first time, and trying to build a simple proof of concept library using tinygo, but getting odd internal LLVM errors. The code seems to build fine with regular big Go.

Code here: [TestLibrary.zip](https://github.com/user-attachments/files/19489066/TestLibrary.zip)

Running `BuildAsReactor.sh` in `TestLibraryWasm` first successfully builds with big Go, and then tries to build with tinygo and fails. It is trying to build to wasm, but quick testing seems to indicate the error persists even when building to native code.

Am I doing something very stupid here, or something unsupported, or is this a bug?

DagAgren, 2025-03-28:
The errors in question are:

```
Call parameter type does not match function signature!
  %cart = call align 4 ptr @runtime.alloc(i32 12, ptr inttoptr (i32 71 to ptr), ptr undef), !dbg !95
 i32  %7 = call i32 @NumberOfCartEntries(ptr %cart), !dbg !102
Incorrect number of arguments passed to called function!
  %7 = call i32 @AddItemToCart(ptr %cart, ptr null, i32 0, i32 %count), !dbg !124
Call parameter type does not match function signature!
  %cart = call align 4 ptr @runtime.alloc(i32 12, ptr inttoptr (i32 71 to ptr), ptr undef), !dbg !140
 i32  call void @RemoveItemFromCart(ptr %cart, i32 %entryIndex), !dbg !148
Call parameter type does not match function signature!
  %cart = call align 4 ptr @runtime.alloc(i32 12, ptr inttoptr (i32 71 to ptr), ptr undef), !dbg !161
 i32  %7 = call i32 @CountOfCartEntry(ptr %cart, i32 %entryIndex), !dbg !169
verification error after compiling package TestLibraryWasm
```

DagAgren, 2025-03-28:
Oh, it seems to be because I used `//export` instead of `//go:wasmexport`, and used unsupported types in the exports.

In that case, it was in fact me doing something stupid, but it might be good to have better error messages here.

aykevl, 2025-03-29:
Normally `//export` doesn't result in these kinds of error messages. Can you share the code that caused these errors?

DagAgren, 2025-03-29:
It's in the linked zip file.

aykevl, 2025-03-31:
Ok, here is the problem.

In TestLibrary/TestLibrary.go, you have this:

```go
//export NumberOfCartEntries
func (cart *Cart) NumberOfCartEntries() int {
    return len(cart.entries)
}
```

And in TestLibraryWasm/Main.go, you have this:

```go
//export NumberOfCartEntries
func NumberOfCartEntries(cartId int) int {
    if cart, ok := carts[cartId]; ok {
        return cart.NumberOfCartEntries()
    }
    return 0
}
```

So you're defining two functions with a different signature but the same name. That is not going to work and indeed can lead to errors like that. (We could maybe detect this particular case and produce a better error message though).

Since you've already solved the issue in a different way, I'm going to close this.

closed 2025-03-31 by aykevl