tinygo-org/tinygo #4848: interp can't handle regexp.MustCompile because it gives up on panic()

dkegel-fastly, 2025-04-09:
interp very sensibly avoids evaluating init functions with side effects.
panic() has the side effect of halting the program.
This means that regexp.MustCompile() is always executed purely at runtime instead of compiletime... leading to slow startup for edge applications, where init runs for every request.

Thinking it through, though, it would be completely reasonable for panic()
inside an init function to halt compilation.  It would give a better
developer experience - invalid regexps in init functions would halt
compilation! - and a better runtime experience, particularly on the edge.

dkegel-fastly, 2025-09-30:
I suppose the benefit of having the panic at build time would go away if the slow compiles were fixed.   We regularly wait two minutes for compiles because of https://github.com/tinygo-org/tinygo/issues/4980 and/or https://github.com/tinygo-org/tinygo/issues/2090.

aykevl, 2025-10-01:
Oh. I see. `interp` is choking on functions with `defer`. That makes sense.

Specifically, I think it's this defer:
https://github.com/golang/go/blob/go1.24.0/src/regexp/syntax/parse.go#L891

This is _probably_ fixable.

davecheney, 2026-11-02:
Rechecked on `dev` at `13a09869`, with Go 1.27.3 and LLVM 23.1.4. `regexp.MustCompile("asdf")` still runs at runtime. I used `//go:noruntime` to check whether initialization was fully evaluated.

On darwin/arm64, plain deferred calls now evaluate at compile time. The remaining blocker here is the deferred `recover()` in `regexp/syntax.parse`. Parsing completes, but `runtime._recover` reads the current runtime task at `src/runtime/panic.go:217`. The interpreter cannot evaluate that read and rolls back the parser call.

This also affects `regexp.Compile("asdf")` and the wrapper that omits the panic. The panic branch in `MustCompile` is not the cause.

On the default `wasip1` target, there are further limits. Even a plain deferred assignment falls back to runtime through `runtime.clearUnwind`. The regexp case rolls back earlier on a load in `regexp/syntax.parse`.

As a control, `-panic-unwind=none` allows `MustCompile("asdf")` to evaluate fully at compile time on both targets. That disables panic recovery, so it is not an equivalent workaround, but it confirms that recovery/unwind support remains relevant to this case.

davecheney, 2026-11-08:
#6138 lets interp run valid regexp initializers at compile time on targets that unwind panics with setjmp. This sample shows it with TinyGo built from dev:

```go
package main

import "regexp"

var re = regexp.MustCompile(`^a+sdf$`)

func main() {
	println(re.MatchString("aaasdf"))
}
```

`regexp/syntax.(*Regexp).Simplify` runs only while a pattern compiles, so it stays in the binary only when the regexp compiles at runtime.

```sh
$ tinygo build -o re . && nm re | grep -c 'regexp/syntax.Regexp).Simplify'
0
$ tinygo build -target wasip1 -scheduler none -o re.wasm . && grep -ac 'regexp/syntax.Regexp).Simplify' re.wasm
0
$ tinygo build -target wasip1 -o re.wasm . && grep -ac 'regexp/syntax.Regexp).Simplify' re.wasm
3
```

Wasm targets that use the asyncify scheduler, which is the wasip1 default, still compile the regexp at runtime, so this issue stays open for that case. Until it is fixed, `-scheduler none` avoids it if the program does not use goroutines.