tinygo-org/tinygo #4860: Support for small IPS USB-C displays like Turing Smart Screen?

tapsu01, 2025-04-17:
Hello! Is there support for Turing Smart Screen? (3.5", 5")

deadprogram, 2025-04-17:
Not at the moment @tapsu01 

What exactly are you wanting to do?

davecheney, 2026-11-08:
@tapsu01 thanks for raising this issue. From what I can tell the Turing screens are some kind of ST7789 style display with a CH552T USB CDC device to bridge them to the USB bus. This is a pretty neat solution because the display owns its frame buffer, and the USB -> SPI bridge allows the host system to update the frame buffer. 

However, a bit of reading on hack-a-day suggests that the onboard micro is a 8051 style system, which is not something tinygo supports, and is likely not flashable.

It would be possible to write replacement control software to drive the display via the USB -> SPI, but that isn't unique to Tinygo.

I'm closing this as answered as I don't think there is a practical path to running tinygo directly on the displays' MCU. Please reopen if you disagree.

closed 2026-11-08 by davecheney