tinygo-org/tinygo #4863: HD44780 LCD Display - Identical .ino and .go code behave differently.

baddda, 2025-04-19:
I was trying to get my setup run with [this](https://github.com/tinygo-org/drivers/tree/release/hd44780) driver. Unfortunately I had several Issues and was able to break it down to the following scenario.

[go file](https://github.com/baddda/tinygo-hd44780-issue/blob/main/main.go)
[ino file](https://github.com/baddda/tinygo-hd44780-issue/blob/main/main.ino/main.ino.ino)

The code mainly executes first some commands to setup the LCD screen. Clears the screen and then sets the cursor to the second row in the second column.

Running the ino file works as expected:

https://github.com/user-attachments/assets/8a45cf82-39cf-4b30-8484-0b351ab6b9e4

Running the go file does not clear the display proper and the cursor is set to row 3 instead of 2.

https://github.com/user-attachments/assets/82184210-82f4-4099-b28e-74baa098349d

The interesting part is the last command of setting the cursor. Based on the [documentation](https://cdn.sparkfun.com/assets/9/5/f/7/b/HD44780.pdf) the command for setting the cursor at row 2 column 2 is `11000001` where the lower bits represent the position in the row. In this case at index 1. With the go file it seems like this command is seen as `11000011`. So positioning the cursor at index 3 (row 4).

davecheney, 2026-10-27:
The reproducer uses D0/D1 for LCD data, which conflicts with TinyGo’s default UART on the Uno.

I suggest using the HD44780 driver’s 4-bit mode, wired to avoid D0/D1. I recently fixed its initialization sequence in [tinygo-org/drivers#928](https://github.com/tinygo-org/drivers/pull/928) and tested it on an Arduino Uno with a 16×2 LCD shield. That change also includes a working example.

Please try the latest drivers code with that wiring and let us know whether it resolves the problem.

davecheney, 2026-11-08:
Thank you for your question. Please feel free to reopen this issue if you have further questions.

closed 2026-11-08 by davecheney