tinygo-org/tinygo #4865: `undefined symbol: os/signal.signal_enable` when compiling with TinyGo WASI target

Mossaka, 2025-04-22:
Hi,

I tried compiling [mcp-go](https://github.com/you/mcp-go) using the latest version of TinyGo with the wasi / wsaip2 target, but ran into the following linker errors:

```console
wasm-ld: error: lto.tmp: undefined symbol: os/signal.signal_enable  
wasm-ld: error: lto.tmp: undefined symbol: os/signal.signal_recv
```

I came across [a similar issue](https://github.com/tinygo-org/tinygo/issues/4250) that was previously closed as resolved, but I’m still hitting this with the current setup. Any guidance or workaround would be much appreciated!

Thanks!

deadprogram, 2025-04-23:
@Mossaka that link to `mcp-go` did not work.

Can you please link to the specific code you were trying to compile and what command you used to compile it with so I can reproduce?

Mossaka, 2025-04-23:
Yes, I was trying to compile this repo: https://github.com/mark3labs/mcp-go

Mossaka, 2025-04-23:
Synced up with @deadprogram online. The code I was trying is `examples/everything`.

dgryski, 2025-05-02:
The `os/signal` package expects those functions to be defined by the runtime.  Currently those functions for tinygo on Linux and Darwin.

dgryski, 2025-05-02:
https://github.com/tinygo-org/tinygo/pull/4882